francescotescari/XiaoMiToolV2 #1: redmi note 7: info_retrive_failed - Failed to get the device unlock token

nicman23, 2019-05-03:
```
[14:43:48][WARN  ][15712ad3] Failed to load system language, switching to English
[14:43:48][ERROR ][15712ad3] Failed to load language: en_US, switching to english
[14:43:48][INFO  ][15712ad3] Starting XiaoMiTool V2 9.4.10 : Linux - amd64 - 5.0.9-arch1-1-bcachefs-00390-gd7dc010d7ea2
[14:43:48][INFO  ][15712ad3] Launching main window
[14:43:50][INFO  ][25e49e8e] Discalimer accepted
[14:43:50][INFO  ][25e49e8e] Action check for updates v2
[14:43:50][INFO  ][25e49e8e] Checking internet connection
[14:43:51][INFO  ][1eaf31f4] xiaomitool.com is reachable
[14:43:51][INFO  ][25e49e8e] Update status: UPDATED
[14:43:51][INFO  ][25e49e8e] Tool is updated
[14:43:51][LOG   ][5ebd1dbc] Feedback token request result: 200 - 02db96a6ef0f4d7e1fe1c56b1f43b421f5d19a06bd2cee1e428518a28121c783e2a0d25f26f4bb81edf2eca385821ffe
[14:43:51][LOG   ][5ebd1dbc] Feedback send result: 200 - 
[14:43:55][WARN  ][9feb625] google.com is not reachable
[14:43:56][INFO  ][9feb625] but it is reachable on port 80
[14:43:56][WARN  ][3f67f167] miui.com is not reachable
[14:43:56][INFO  ][3f67f167] but it is reachable on port 80
[14:44:06][PSTA  ][25e49e8e] Start process (1): "/home/nikos/Downloads/New Folder/opt/xiaomitool/res/tools/adb" "kill-server"
[14:44:06][INFO  ][25e49e8e] Process (1) ended with exit code: 0, output len: 0
[14:44:06][PSTA  ][25e49e8e] Start process (2): "/home/nikos/Downloads/New Folder/opt/xiaomitool/res/tools/adb" "start-server"
[14:44:06][PROC  ][48e8b14d] Process (2) output: * daemon not running; starting now at tcp:5037
[14:44:09][PROC  ][48e8b14d] Process (2) output: * daemon started successfully
[14:44:09][INFO  ][25e49e8e] Process (2) ended with exit code: 0, output len: 76
[14:44:09][PSTA  ][d8f9900] Start process (3): "/home/nikos/Downloads/New Folder/opt/xiaomitool/res/tools/adb" "track-devices"
[14:44:09][INFO  ][25e49e8e] Searching at least one device
[14:44:09][PSTA  ][42917958] Start process (4): "/home/nikos/Downloads/New Folder/opt/xiaomitool/res/tools/adb" "devices"
[14:44:09][PROC  ][2af8cd19] Process (3) output: 000fc0a7856	device
[14:44:09][PROC  ][63b83e7b] Process (4) output: List of devices attached
[14:44:09][PROC  ][63b83e7b] Process (4) output: c0a7856	device
[14:44:09][PROC  ][63b83e7b] Process (4) output: 
[14:44:09][INFO  ][42917958] Process (4) ended with exit code: 0, output len: 40
[14:44:09][PSTA  ][42917958] Start process (5): "/home/nikos/Downloads/New Folder/opt/xiaomitool/res/tools/fastboot" "devices"
[14:44:09][INFO  ][42917958] Process (5) ended with exit code: 0, output len: 0
[14:44:09][INFO  ][42917958] Connected device: c0a7856 -> device
[14:44:09][PSTA  ][42917958] Start process (6): "/home/nikos/Downloads/New Folder/opt/xiaomitool/res/tools/adb" "-s" "c0a7856" "shell" "-x" "getprop"
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [DEVICE_PROVISIONED]: [1]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [af.fast_track_multiplier]: [1]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [audio.deep_buffer.media]: [true]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [audio.offload.min.duration.secs]: [30]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [audio.offload.video]: [true]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [bt.max.hfpclient.connections]: [1]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [camera.aux.packagelist]: [org.codeaurora.snapcam,com.android.camera,com.qualcomm.qti.qmmi,com.longcheertel.cit]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.appimageformat]: [lz4]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.bg-dex2oat-threads]: [2]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.boot-dex2oat-threads]: [6]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.dex2oat-Xms]: [64m]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.dex2oat-Xmx]: [512m]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.dex2oat-minidebuginfo]: [true]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.dex2oat-threads]: [6]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.dexopt.secondary]: [true]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.heapgrowthlimit]: [256m]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.heapmaxfree]: [8m]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.heapminfree]: [512k]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.heapsize]: [512m]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.heapstartsize]: [8m]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.heaptargetutilization]: [0.75]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.image-dex2oat-Xms]: [64m]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.image-dex2oat-Xmx]: [64m]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.isa.arm.features]: [default]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.isa.arm.variant]: [cortex-a73]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.isa.arm64.features]: [default]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.isa.arm64.variant]: [generic]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.stack-trace-dir]: [/data/anr]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.usejit]: [true]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [dalvik.vm.usejitprofiles]: [true]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [debug.atrace.tags.enableflags]: [0]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [debug.force_rtl]: [0]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [debug.gralloc.enable_fb_ubwc]: [0]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [debug.gralloc.gfx_ubwc_disable]: [1]
[14:44:09][PROC  ][2d4afcc1] Process (6) output: [debug.sf.disable_backpressure]: [1]
[... 2377 lines cut ...]
[14:47:15][INFO  ][42917958] Process (195) ended with exit code: 0, output len: 16
[14:47:15][INFO  ][42917958] Connected device: c0a7856 -> fastboot
[14:47:17][PSTA  ][42917958] Start process (196): "/home/nikos/Downloads/New Folder/opt/xiaomitool/res/tools/adb" "devices"
[14:47:17][PROC  ][538a5bb8] Process (196) output: List of devices attached
[14:47:17][PROC  ][538a5bb8] Process (196) output: 
[14:47:17][INFO  ][42917958] Process (196) ended with exit code: 0, output len: 25
[14:47:17][PSTA  ][42917958] Start process (197): "/home/nikos/Downloads/New Folder/opt/xiaomitool/res/tools/fastboot" "devices"
[14:47:17][PROC  ][77b7ff98] Process (197) output: c0a7856	fastboot
[14:47:17][INFO  ][42917958] Process (197) ended with exit code: 0, output len: 16
[14:47:17][INFO  ][42917958] Connected device: c0a7856 -> fastboot
[14:47:19][LOG   ][5221a77a] Feedback token request result: 200 - 884194d063664ac53f6a078cf563927eba5ca38e904928c551b3a5b4ee8b542ca97d488b01326dee2befb837c43ef9b3
[14:47:19][LOG   ][5221a77a] Sending feedback with token: ba5ca38e904928c551b3a5b4ee8b542ca97d488b01326dee2befb837c43ef9b3
[14:47:19][LOG   ][5221a77a] Feedback send result: 200 - 
[14:47:19][LOG   ][5b116252] Feedback token request result: 200 - 02db96a6ef0f4d7e1fe1c56b1f43b421b4f1cf5c19e9304e02e1504233365ffe344e21c0a0e9c88d3ae07426d73ebd9e
[14:47:20][PSTA  ][42917958] Start process (198): "/home/nikos/Downloads/New Folder/opt/xiaomitool/res/tools/adb" "devices"
[14:47:20][PROC  ][39e8b952] Process (198) output: List of devices attached
[14:47:20][PROC  ][39e8b952] Process (198) output: 
[14:47:20][INFO  ][42917958] Process (198) ended with exit code: 0, output len: 25
[14:47:20][PSTA  ][42917958] Start process (199): "/home/nikos/Downloads/New Folder/opt/xiaomitool/res/tools/fastboot" "devices"
[14:47:20][PROC  ][3f6e43f2] Process (199) output: c0a7856	fastboot
[14:47:20][INFO  ][42917958] Process (199) ended with exit code: 0, output len: 16
[14:47:20][INFO  ][42917958] Connected device: c0a7856 -> fastboot
[14:47:20][LOG   ][5b116252] Feedback send result: 200 - 

```

hannay1, 2019-07-11:
I have the same issue it seems. Running MiUi 10.3.6.0 global on a Pocophone f1, flashing from debian based system, 95% sure I am not subject to the 72 hour unlock window

DDvO, 2021-12-12:
A also get "failed to get device unlock token" for a Xiaomi Redmi Note 8 2021.
I had logged in with a Xiaomi account on the phone before trying to unlock.
I tried with an account that had been active for about 3 months, and with another account I created yesterday.

Dipaarif, 2022-03-03:
A

pyramid3d, 2022-05-30:
Running ArchLinux.
Did get "failed to get the device unlock token" for a Xiaomi Redmi Note 8 2021.
code: info_retrieve_failed (origin:com.xiomitool.v2.procedure.install.FasybootInstall$4.run(Unknown Source)
I had logged in with a Xiaomi account on the phone before trying to unlock. Did not factory reset or data wipe.
I tried with a Xiaomi account that had been active for about 1 month and on the device registration of phone to account was successful.