francescotescari/XiaoMiToolV2 #3: Release for Non-Debian Distros

Abhinav1217, 2019-08-20:
Hello, Your recent release is windows only. I am sure you will provide latest build for linux when you get time. Can we hope you provide build for Non-Debian distros. At least have support for universal package like snap (easy to convert .deb to snap ), flatpak ( little complicated ), appimage or old time favourite, .tar.gz. 

These three universal package are supported on almost all linux varient from arch, solus, fedora, gentoo etc. It would be nice for people like me who uses linux which is not ubuntu or other debian derivative.

MatusGoc, 2020-05-02:
If you're on an Arch based distro, you can get it on the AUR.
https://aur.archlinux.org/packages/xiaomitool-v2

But a .tar.gz release would be nice.

uAliFurkanY, 2021-07-23:
snap is nowhere near universal right now. It's just a containerized package distribution system for Ubuntu.

Flatpak and AppImage on the other hand, are universal and easy to install on every distro.