francescotescari/XiaoMiToolV2 #13: Linux fastboot error

MBARESERRA, 2020-11-15:
after i select the device in xmt, the phone reboot in fastboot but xmt doesn't find any device.

flozzone, 2021-01-01:
You could try to install the`fastboot` package of your Linux distribution, which should install the required udev rules for the required USB devices to work properly with fastboot. After installing you should make sure new rules are loaded with `udevadm control --reload-rules && udevadm trigger` and unplug and replug your device.

fenuks, 2021-02-24:
I have the same problem. USB debugging is enabled, OEM unlocking as well. XMT reboots phone into fastboot mode, and afterwards cannot detect it, although by running `fastboot devices` I can see it. I'm using Arch Linux, and `android-tools` package from community repository.

0xSteeW, 2021-04-03:
I will add my two cents: fastboot only detects my devices if running as root.