francescotescari/XiaoMiToolV2 #14: App is corrupted in Mac OS Big Sur

Olivier-SGO, 2020-11-21:
Can't open the last version in Mac OS, says the app is corrupted and I should put it in trash, whereas the previous version says I don't have the authorisation to open the app.... stuck

moskalukigor, 2020-12-05:
> Can't open the last version in Mac OS, says the app is corrupted and I should put it in trash, whereas the previous version says I don't have the authorisation to open the app.... stuck

Open Terminal and execute: `xattr -rd com.apple.quarantine <app>`
Example: `xattr -rd com.apple.quarantine /Users/USERNAME/Downloads/XiaoMiToolV2.app`

kyle-import, 2021-02-20:
> Open Terminal and execute: `xattr -rd com.apple.quarantine <app>`
> Example: `xattr -rd com.apple.quarantine /Users/USERNAME/Downloads/XiaoMiToolV2.app`

This isn't a solution. The app is corrupted, it won't load.

tupac19861013, 2021-04-17:
Has anyone found a way to open this application?

moskalukigor, 2021-04-20:
> This isn't a solution. The app is corrupted, it won't load.

But it works great. Why isn't this a solution?
@tupac19861013 you can try my solution

Rasinari, 2021-06-02:
> > This isn't a solution. The app is corrupted, it won't load.
> 
> But it works great. Why isn't this a solution?
> @tupac19861013 you can try my solution

xattr no longer works on Big Sur

chetanpm, 2021-06-05:
I was wondering how to solve the damaged application, xattr works for me on Big Sur 11.2.3