francescotescari/XiaoMiToolV2 #17: Invalid fastboot token

tsoernes, 2021-03-22:
```
Failed to unlock your device, Xiaomi server returned error 20044:
Error description: Unknown Error: 20044
Server description: Invalid fastboot token: Please retry or upgrade your tool
```
I'm using the latest version. What could be the cause of this?

GribanovIvan, 2021-04-19:
Maybe need wait for week or something.. Solved now?

tsoernes, 2021-04-20:
yes!

GribanovIvan, 2021-04-20:
> yes!
> […](#)

Thanks for reply, now you can close issue and devs get info about error code

closed 2021-04-20 by tsoernes

asajadi84, 2021-06-28:
This exact message showed up to me as well, any clue how long I have to wait? (as you know the official Mi Unlock tool tells user the remaining wait time in hours.)

Uphanjoel, 2023-01-05:
> yes!
> […](#)

is there any fix that I can use ?