francescotescari/XiaoMiToolV2 #23: Xiaomi procedure failed: [getServiceToken] Missing serviceToken cookie

erikrodrgz, 2021-07-13:
I followed the instructions to unlock my Xiaomi phone, but I found this procedure error in the current linux version and in the Windows version.

[image]

FrosT2k5, 2021-07-13:
did u find any solution for it so far?

erikrodrgz, 2021-07-13:
> did u find any solution for it so far?

No, do you reproduce this issue?

FrosT2k5, 2021-07-13:
Yes dude I am trying to solve my friends blocked bootloader in linux, but this doesnt seem to work sad. You got any other working way?

erikrodrgz, 2021-07-13:
> Yes I did dude. I am trying to solve my friends blocked bootloader in linux, but this doesnt seem to work sad. You got any other working way?

Unfortunately no, the closest I got was through Xiaomitoolv2. I tried with the official tool in Win10, but the line with the USB connection just blinked once and ejected the USB connection. I do not try yet to format the cell phone to an old android and then try again. But I think it's possible a server problem, I don't know, maybe I will wait sometime to try again.

FrosT2k5, 2021-07-14:
> Unfortunately no, the closest I got was through Xiaomitoolv2. I tried with the official tool in Win10, but the line with the USB connection just blinked once and ejected the USB connection. I do not [...]

That's sad... I just made him unlock with official tool in windows. It worked it seems, nice

fengzee, 2021-07-14:
I got stuck at the same issue. I think Xiaomi has just made changes to their service and requiring a "serviceToken" for server authentication. I have just 1 more day left before I can unlock my phone. Sad. I don't have a Windows device so I cannot use the Xiaomi official tool. Planning to find an internet cafe with Windows PCs.

FrosT2k5, 2021-07-14:
> I got stuck at the same issue. I think Xiaomi has just made changes to their service and requiring a "serviceToken" for server authentication. I have just 1 more day left before I can unlock my [...]

U use Linux? Maybe try using windows using kvm. Maybe that will work. Idk a shit about it tho haha. Hope u find someway. Do let me know what u do

Usaamah99, 2021-07-14:
I'm getting that too on a Mac

swarupbhanjachowdhury, 2021-07-15:
Same issue on Linux

FrosT2k5, 2021-07-15:
No solution/workaround for it as of now... Sad. Maybe we should as for Linux tool in miui forum. They say people who need unlocked for Linux are very less

fengzee, 2021-07-15:
> U use Linux? Maybe try using windows using kvm. Maybe that will work. Idk a shit about it tho haha. Hope u find someway. Do let me know what u do

I'm a Mac user. I have virtual machine options on Mac (eg. Parallel Desktop) too but I'm too lazy to download a Windows system image. I'm pretty sure most Xiaomi engineers also work on Macs. It's insane they only provide a Windows version of their tool.

FrosT2k5, 2021-07-16:
> I'm a Mac user. I have virtual machine options on Mac (eg. Parallel Desktop) too but I'm too lazy to download a Windows system image. I'm pretty sure most Xiaomi engineers also work on Macs. It's [...]

Yeah man sad. Why they wanna do all this, permission and all. just let fastboot oem unlock, unlock the bootloader, it'll be simple af and can run it anywhere. Even on phone using bugjaegar

DTFxDeadlock, 2021-07-17:
Facing same problem in my laptop.pls solve the problem as fast as possible .I need that in argent
[image]

higorhi, 2021-07-21:
I'm a Windows user, had the same issue of unrecognizable device in fastboot. Flashed a live boot Linux USB drive with Solus OS, got XiaoMiToolV2 working and all, did all the `udev` shenanigans to allow device access permissions, and got stuck on the same issue. Seeing as this is happening on all three platforms and that all reports of that are all within the last week, I'm highly suspecting of Xiaomi's services instability, really.

I don't know if you have had this other issue as well, but when trying to sync the account to allow for bootloader unlocking, it now fails basically all the time. I think I was lucky on Sunday and got a success response once but that was it; all other responses are something of the likes of "Couldn't execute sync procedure. Please try again".

This behavior makes me think the issues are somehow related. I'll keep you guys posted.

CavaleriOmar, 2021-07-22:
I have been trying to unlock my, now old, Redmi 6 for three days to install a custom rom ... but the error exposed by erikrodrgz keeps happening to me.

What mistake is that anyway?

FrosT2k5, 2021-07-22:
> I have been trying to unlock my, now old, Redmi 6 for three days to install a custom rom ... but the error exposed by @erikrodrgz keeps happening to me.
> 
> What mistake is that anyway?
> 
> PS: I'm on Linux

It's not your mistake, its xiaomi messed something up I think, right now solution is somehow get windows and use the xiaomi's official unlocker...

kagiannis, 2021-07-22:
I had the same problem in Linux. I was able to do it in windows with the official app. But I needed to run `bcdedit /set testsigning on` as administrator and then run XiaomiToolv2 to install some drivers needed.

mconrado, 2021-07-22:
I have same problem. Probably the Xiaomi site change something

DavidHenryThoreau, 2021-07-22:
> I had the same problem in Linux. I finally did it in windows with the official app. But I needed to run `bcdedit /set testsigning on` as administrator and then run XiaomiToolv2 to install some needed drivers .

The phone (Redmi Note 8T) does not appear in the official software

uAliFurkanY, 2021-07-23:
Same with Linux and Windows (kvm + usb redirection)

CavaleriOmar, 2021-07-25:
@uAliFurkanY how did you set up the USB connection?

Using kvm I was unable to get xiaomi recognized not only by XMT2, but not even by MiPC suite... Windows recognizes it, bit Mi softs no...

kagiannis, 2021-07-25:
@CavaleriOmar  in qemu you can do this `-usb -device usb-host,hostbus=NUM,hostport=NUM` where you can find the numbers with `lsusb`. But it didn't work for me. I don't know if it is a problem with my laptop as also fastboot had problems. Only using a different laptop with windows 10 it worked.

uAliFurkanY, 2021-07-25:
I was using virt-manager so I just redirected the USB debice every reboot.

Mi's official tool worked in the end, I'm typing from DotOS.

CavaleriOmar, 2021-07-26:
@uAliFurkanY , same VM

Done.

Windows 10 under VM... MI PC suite did not recognize it simply because the Redmi 6 is out of support ...

MI Flash Unlock found it right away (yesterday it didn't recognize it because I'm a jerk and I hadn't put it in fastboot).

pixrafal, 2021-07-30:
I got same problem, too. 
I had been used it to unlock/twrp/flash many Mi-devices, included Redmi Note5/Note7/Note8/K20/K30/K30s, on my mac (Mojave 10.14.6) without any issue, but recently when I tried to unlock my new Redmi 9T , I got same message and always failed. even I tried to check old success unlock/flash devices, I got same message, too.
 
I don't have any PC with win7/10, and I also don't use VM with win10.

did MIUI server change rules?

ahmadraniri1994, 2021-07-31:
@pixrafal  wow, world seems small huh? I'm facing same issue with Redmi 9T, fortunately I seem not alone. Keep monitor this issue guys.

FrosT2k5, 2021-07-31:
> I got same problem, too.
> I had been used it to unlock/twrp/flash many Mi-devices, included Redmi Note5/Note7/Note8/K20/K30/K30s, on my mac (Mojave 10.14.6) without any issue, but recently when I [...]

Yes seems like Miui server changed something. Sad... This is the worst thing about xiaomi

vardhamann, 2021-08-07:
Having the same issue!. hopefully, this gets resolved soon.

AlecsFerra, 2021-08-10:
Same issue probably xiaomi changes his APIs

marrobHD, 2021-08-11:
Even if this tool would work, you still have to wait because this unlock tool is using Xiaomis APIs.

7iomka, 2021-08-11:
[image]
Error when I'm try to unlock locked bootloader on xiaomi redmi note 7 pro prime (after upgrading to miui 12.5 all google services not works and I need root for go back to miui 11..)

yxles-dev, 2021-08-11:
> Even if this tool would work, you still have to wait because this unlock tool is using Xiaomis APIs.

Oh I thought it wouldnt because I found this on a video that says you dont need to wait and instantly unlock the bootloader of your device. Oh well, i just did a check and found out that my device had already unlocked bootloader so I will only use the program for flashing custom roms

CavaleriOmar, 2021-08-12:
>Oh, pensavo che non lo sarebbe stato perché l'ho trovato su un video che dice che non è necessario attendere e sbloccare immediatamente il bootloader del dispositivo. Vabbè, ho appena fatto un controllo e ho scoperto che il mio dispositivo aveva già sbloccato il bootloader quindi userò il programma solo per flashare le custom rom

Dà problemi pure nel flashing, perché prima di flashare una custom rom devi flashare TWRP, e la funzione specifica dà sempre errore... e provare a flashare una custom rom DOPO aver flashato in autonomia TWRP... beh, dà sempre errore.

Praticamente ora è inutilizzabile come programma.

dessalines, 2021-08-15:
Same issue here.

vmavromatis, 2021-08-16:
Not sure if it's of any help, just to report that same issue happening here Linux + Redmi 5 Plus

morfikov, 2021-08-20:
I also have the exact same issue with my Xiaomi Redmi 9 (galahad)

KevinSleegers, 2021-08-20:
Having the same issue with my Poco F3.

tomyo, 2021-08-22:
Same issue on arch linux with redmi note 10 pro.

LuuisBeck, 2021-08-23:
Same problem here, Redmi Note 5 with macOS Catalina

Nikitian, 2021-08-23:
Have this problem too. Bug Sur + Mi 10

vmavromatis, 2021-08-23:
guys, has anyone been able to build the project from the linux branch and re-package it for the rest of us?
maybe it will work because it's 1 commit ahead of master........

d3417, 2021-08-23:
You need to change the Login URI to **https://account.xiaomi.com/pass/serviceLogin?sid=unlockApi&json=false&passive=true&hidden=false&_snsDefault=facebook&_locale=en&checkSafePhone=true** and make Phone SMS Verification, after you get the correct cookie for Unlock your devices.

\ò/

d3417, 2021-08-23:
`git clone https://github.com/francescotescari/XiaoMiToolV2`

**src/main/java/com/xiaomitool/v2/gui/controller/LoginController.java** Line 39:

Replace that line with:
 `   private static final String LOGIN_URL = "https://account.xiaomi.com/pass/serviceLogin?sid=unlockApi&json=false&passive=true&hidden=false&_snsDefault=facebook&checkSafePhone=true&&_locale=" + Locale.getDefault().getLanguage().toLowerCase();
`
After that do `gradle build && gradle run` , login with 2FA verification, try to unlock your bootloader!

I've not tried myself because i don't want to use another smartphone for 2FA Verification and to Lock & ReUnlock the Bootloader, but if works, please mention me in this thread and tell me if you're able to unlock bootloader and mod your Xiaomi Smartphone!

d3417, 2021-08-23:
**https://mega.nz/file/4UAjgKrC#Fk1V172GES7M4IUpg_pCfJkGVYXTBOSKxvIj93kuMeA**

Just extract and `bash run.sh`

Happy modding!

vmavromatis, 2021-08-23:
> **https://mega.nz/file/4UAjgKrC#Fk1V172GES7M4IUpg_pCfJkGVYXTBOSKxvIj93kuMeA**
> 
> Just extract and `bash run.sh`
> 
> Happy modding!

The file you are trying to download is no longer available.
This link is unavailable as the user’s account has been closed for gross violation of MEGA’s Terms of Service.
:( can you upload somewhere else?

morfikov, 2021-08-23:
> `git clone https://github.com/francescotescari/XiaoMiToolV2`
>
> **src/main/java/com/xiaomitool/v2/gui/controller/LoginController.java** Line 39:
>
> Replace that line with:
> ` private static final String [...]

I tried to do that:

```
$ git clone https://github.com/francescotescari/XiaoMiToolV2                                                               
Cloning into 'XiaoMiToolV2'...
remote: Enumerating objects: 2569, done.
remote: Counting objects: 100% (2566/2566), done.
remote: Compressing objects: 100% (778/778), done.
Receiving objects: 100% (2569/2569), 67.78 MiB | 388.00 KiB/s, done.
remote: Total 2569 (delta 1593), reused 2566 (delta 1593), pack-reused 3
Resolving deltas: 100% (1593/1593), done.
Updating files: 100% (332/332), done.
                                                                                                                                     
╭─[morfik@morfikownia] - [/media/debuilder/git-xiaomimitoolv2] - [2021-08-23 23:16:37]
╰─[0] < > $  cd XiaoMiToolV2
                                                                                                                                     
╭─[morfik@morfikownia] - [/media/debuilder/git-xiaomimitoolv2/XiaoMiToolV2] - [2021-08-23 23:20:27]
╰─[0] < > $  geany src/main/java/com/xiaomitool/v2/gui/controller/LoginController.java 

╭─[morfik@morfikownia] - [/media/debuilder/git-xiaomimitoolv2/XiaoMiToolV2] - [2021-08-23 23:20:47]
╰─[0] < > $  gradle build && gradle run
Starting a Gradle Daemon (subsequent builds will be faster)

> Task :buildEnvironment

------------------------------------------------------------
Root project
------------------------------------------------------------

classpath
No dependencies

BUILD SUCCESSFUL in 12s
1 actionable task: 1 executed

FAILURE: Build failed with an exception.

* What went wrong:
Task 'run' not found in root project 'XiaoMiToolV2'.

* Try:
Run gradle tasks to get a list of available tasks. Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1s
```

Should I do something else to make it work?

d3417, 2021-08-23:
> I tried to do that:
>
> ```
> $ git clone https://github.com/francescotescari/XiaoMiToolV2
> Cloning into 'XiaoMiToolV2'...
> remote: [...]

Wait few seconds, i'm reuploading it on DropBox.com!

morfikov, 2021-08-23:
Ok.

d3417, 2021-08-24:
> Ok.

**https://dubox.com/s/1RLWtwwYzoz6W3R4INjyKbw** 
Password: **nvrs**

`unzip XiaoMiToolV2(Alpha Version by Denis Sossich).zip
cd "XiaoMiToolV2(Alpha Version by Denis Sossich)" 
bash run.sh`

d3417, 2021-08-24:
> > **https://mega.nz/file/4UAjgKrC#Fk1V172GES7M4IUpg_pCfJkGVYXTBOSKxvIj93kuMeA**
> > Just extract and `bash run.sh`
> > Happy modding!
> 
> ```
> The file you are trying to download is no longer available.
> This link is unavailable as the user’s account has been closed for gross violation of MEGA’s Terms of Service.
> ```
> 
> :( can you upload somewhere else?
> 
> PS: I will open a PR for this

**https://dubox.com/s/1RLWtwwYzoz6W3R4INjyKbw**
Password: **nvrs**

`unzip XiaoMiToolV2(Alpha Version by Denis Sossich).zip`

`cd "XiaoMiToolV2(Alpha Version by Denis Sossich)"`

`bash run.sh`

d3417, 2021-08-24:
> ```
> The file you are trying to download is no longer available.
> This link is unavailable as the user’s account has been closed for gross violation of MEGA’s Terms of Service.
> ```
>
> :( can you upload [...]

Thank you very much for correcting the obsolete code by integrating my URI of Xiaomi with the Login 2FA, now all that remains is to try a BootLoader Unlock and see if it works as I predicted, I based on MiUnlockTool that has in a folder a file called app-config, in this file there is the Login URI that integrates the 2FA, just entered the Code via SMS in the Login Box, the App MiUnlockTool registers the user and allows the bootloader unlock.

If I've solved the problem, which is at the root of the software's operation (If it doesn't unlock the Bootloader it can't perform other actions), I'd be pleased to appear in the Credits of Contributing Developers with the name "Denis Sossich".

morfikov, 2021-08-24:
Ok, I've managed to figure out what was wrong with my setup. Gradle version (4.4.1) was to old for the java version (11.0.12) that my system was using. I had to update gradle to 7.2. Also gradle wrapper needed to be updated, but it updated automatically. Then it worked, but the linux adb/fastboot tools were missing (only exe were in the github dir). I copied the whole res/ dir from XMT2_Linux_20.7.28.run to the github source. Then after running `gradlew run` the GUI showed and recognized my phone. I selected the unlock the bootloader option, and it looks it works:

https://i.imgur.com/TpJs8UU.png

I still have to wait 3 days, but I think it will work just fine now. :]

d3417, 2021-08-24:
> Ok, I've managed to figure out what was wrong with my setup. Gradle version (4.4.1) was to old for the java version (11.0.12) that my system was using. I had to update gradle to 7.2. Also gradle [...]

Perfect, my piece of code it's a fix! I'm happy for this ^_^

CavaleriOmar, 2021-08-24:
@d3417 Congrats on the accomplishment... I'm still considering whether to get another XIaomi to replace my Redmi 6 that's ready to abandon me... this may weigh on my choice

pixrafal, 2021-08-25:
@d3417 thanks a lot, I tested it and looks like it pass 2FA verification, but I got failed message on terminal :

[15:22:40][PSTA  ][-------] Start process (826): "./res/tools/fastboot" "-s" "xxxxxxxx: uid or device id" "oem" "unlock" "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx.....: huge numbers, maybe it's token or private key?"
[15:08:14][PROC  ][][-------]] Process (826) output:                                                    FAILED (Command length to RawCommand() is too long)
[15:22:40][PROC  ][-------] Process (826) output: fastboot: error: Command failed
[15:22:40][INFO  ][-------] Process (826) ended with exit code: 1, output len: 134
[15:22:40][WARN  ][-------] InstallException created: unlock_error - Failed to unlock the device, fastboot exit with status non zero or internal error

here is error message from app
[image]

I did try a lot and get same result, and can not receive 2FA code today.

andresangelini, 2021-08-27:
Good news! Thanks to @morfikov's [post](https://morfikov.github.io/post/jak-odblokowac-bootloader-w-xiaomi-redmi-9-galahad-lancelot/) I figured out that perhaps I should try installing the Java Development Kit:
```
sudo apt install default-jdk
```
Alas, after firing XiaoMiToolV2 again:
```
./gradlew run
```
Now I could finally advance to the next stage (choosing whether your phone is working normally or it's bricked).

I'll report back again soon.

andresangelini, 2021-08-28:
Ok. I just bumped with a new problem (possible two but I think they are related).
Since this the device that I'm using for 2FA, I try to log in into my Xiaomi account through XiaoMiToolV2 first, before doing anything to it. However, when I enter my 2FA code sent via SMS, it doesn't recognized it as valid. I eventually was able to bypass this by momentarily disabling 2FA through the Xiaomi website but then, when trying to ask for the authorization token, XiaoMiToolV2 says that Xiaomi servers returned a 20031 error: "This device is not linked to an account" (even though it has been linked since I bought it years ago).

Any ideas? Thanks once again.

morfikov, 2021-08-28:
When you log into your account on the website (https://account.xiaomi.com/), do you see your phone listed there?

andresangelini, 2021-08-28:
> When you log into your account on the website (https://account.xiaomi.com/), do you see your phone listed there?

Yes, I do see it there. Actually, I don't recall having this issue with the XMT2_Linux_20.7.28.run file, so perhaps the key lays on the link we replaced in 39 of src/main/java/com/xiaomitool/v2/gui/controller/LoginController.java. I'll try to look a little bit more into it tomorrow since I almost ran out of 2FA trials for today.

It may take [some time]( https://github.com/francescotescari/XiaoMiToolV2/pull/26) until @francescotescari checks out and accept the @vmavromatis' pull request but I'm hoping to fix the problem I'm having anyways just in case anyone is going through the same and doesn't want to wait until a patched version come out.

3nuc, 2021-08-28:
> @andresangelini I've written an article on how I've managed to unlock my phone. [Take a look at it](https://morfikov.github.io/post/jak-odblokowac-bootloader-w-xiaomi-redmi-9-galahad-lancelot/#xiaomi-procedure-failed-getservicetoken-missing-servicetoken-cookie). It's in Polish, but all the necessary commands are included for ctrl+c and ctrl+v. :)

Thanks! Following these instructions helped me unlock by bootloader. I couldn't unlock it using the official https://en.miui.com/unlock/download_en.html program (I got a "Your account is not authorized for this operation" error and didn't know what to do), but applying the patch the switching to the "linux" branch did it for me :)

[194 more replies]

closed 2021-10-05 by francescotescari