francescotescari/XiaoMiToolV2 #25: Cannot find a rom that can recover your device error.

Trakwold, 2021-08-10:
Tool :
    Version   : 20.7.28 (Beta)
    Host OS : Manjaro Arch Linux KDE Plasma
Phone : 
    name : Redmi Note 7 
    code  : Lavender
    serial : a67b51f

Error :
[image]

Last step :
[image]

My device is bricked with MIUI 12.5.1.0.
My device has always ran Global. The program only looks for Chinese rom. Is there any way to choose a rom on my local hardrive? Anyway thanks for reading.

AlainnnLN, 2021-12-17:
hi can you resolve this problem ? cause i have it !

neppedboy, 2021-12-29:
Well, i have simmilar problem but with Xiaomi Redmi Note 8 Pro. Program just don't get an info about device, I should type it manually

kytkaprogramuje, 2022-01-30:
You saying that you are on linux using this tool but i think that the tool is only for windows. And if you using wine it could mean that its not gonna work.

 (sorry for bad english)