francescotescari/XiaoMiToolV2 #46: Xiaomi procedure failed: [getServiceToken] Missing serviceToken cookie. On windows 10

babydickdanilko, 2021-12-04:
[image]

This error appeared on Windows 10, Xiaomi Redmi 10 (not note), found a similar thread, but there is about Linux

itsMustafa, 2021-12-05:
Same error on Linux as well
OS: Ubuntu 21.10

hcarrega, 2021-12-11:
Same on macOS Monterey

HellBringer419, 2021-12-15:
Same on Linux (Manjaro KDE).
Trying to use Lineage-microG on Redmi Note 8 (gingko)

barthmania, 2021-12-27:
Same error for me on MacOS Monterey.

joergrs, 2022-03-11:
I got the same problem. But I was able to unlock my phone (Redmi Note 10 Pro) using the official Mi Unlock tool 6.5 from https://en.miui.com/unlock/download_en.html

Before I tried Mi Unlock 5.5 which also gave me errors. So my uneducated guess is Xiaomi changed some of their APIs.