francescotescari/XiaoMiToolV2 #48: Cannot find any rom that can recover your device

Badrakh25, 2021-12-06:
[image]
it seems like cannot fetch chinese roms maybe sources endpoint changed or something

AnonymousWebHacker, 2022-09-08:
Cuando buscas tu modelo, te salen varios nombres , ejemplo (dandelion dev, dandelion new, dandelion, dandelion bug ..etc) , selecciona simplemente el codename de tu dispositivo, "dandelion" y descargara la rom para el.

Revisa en ajustes, y espesifica antes, que region quieres

156Strix, 2022-09-09:
i have the same problem. and the tool keeps trying to load chinese rom even though i choose global.

AnonymousWebHacker, 2022-09-10:
> i have the same problem. and the tool keeps trying to load chinese rom even though i choose global.

But if your device is bricked, it doesn't matter what rom you install, what you need is to install an operating system, later on, you change to another system with Global, Russian, etc.

156Strix, 2022-09-10:
> > i have the same problem. and the tool keeps trying to load chinese rom even though i choose global.
> 
> But if your device is bricked, it doesn't matter what rom you install, what you need is to install an operating system, later on, you change to another system with Global, Russian, etc.

my problem is i can't install any operating system

AnonymousWebHacker, 2022-09-10:
> my problem is i can't install any operating system

https://t.me/MIUISPAIN