francescotescari/XiaoMiToolV2 #50: Xiaomi procedure failed: [getServiceToken] Missing serviceToken cookie on mac M1

id3vz, 2021-12-17:
I faced this error "Xiaomi procedure failed: [getServiceToken] Missing serviceToken cookie" on m1 MacBook pro
Anyone knows how to build it on m1 mac so I can use the last commit in the build

hcarrega, 2021-12-18:
same error where, i dont know if that causa my 168 hours extension again

lucaslira95, 2021-12-28:
I'm having the same problem on linux.
Did you guys managed to solve the problem?

hcarrega, 2021-12-28:
Maybe something's is missing . Even try flashing magisk via app or install twrp I get the same error

TadaCZE, 2022-01-01:
### PROBLEM SOLVED!
The problem is that the version available for download on the official website and the release page is too old! The problem has been patched but you have to build it yourself using the following commands:

`git clone https://github.com/francescotescari/XiaoMiToolV2`
`./gradlew build`
`./gradlew run`

On some computers this may not work (my desktop PC works, but my laptop failed to build), so in that case try a VM and copy over the built files. :+1:

lucaslira95, 2022-01-02:
> ### PROBLEM SOLVED!
> The problem is that the version available for download on the official website and the release page is too old! The problem has been patched but you have to build it yourself [...]

Build fails in my laptop and desktop, can you share the compiled binary?

hariantara, 2022-01-05:
## For someone use the Macos Intel you can replicate my way (Make sure your Android JDK is setup first or you can install Android Studio to get All Android Setting)

1. Clone the master repo ```git clone https://github.com/francescotescari/XiaoMiToolV2```
2. After clone open Code Editor (I Use VSCode), and open your terminal
3. In Terminal go to XMT tool project folder where you keep in
4. After go to XMT directory in your terminal, please fetch Mac Branch by using this command
    * ```git fetch mac```
    * after that copy file and keep in another folder  of ```adb``` and ```fastboot```
    * after that, in your terminal use this command ```git fetch master``` and replace all file inside the ```./res/tools/``` with ```adb``` and ```fastboot```
    * after that ```sudo ./gradlew build``` and then ```sudo ./gradlew run```
5. XMT will be open and do what you need

## Note: XMT that works in 2022 is, when you login mi account in XMT it request SMS vefirication, if your XMT not request SMS verification when login, it is not updated

Hasodikis, 2022-01-06:
Please guys ....!!!! Please could you provide us with a binary...
No matter what I try ...I keeps on failing

swayok, 2022-01-25:
Thank you! this worked!
Note: you need to have exactly OpenJDK 11 (https://jdk.java.net/archive/). I used 64 bit version 11.0.2 (build 11.0.2+9)
On windows you may also need to add/modify environment variables:
```
JAVA_HOME -> S:\JDK\11
PATH -> Edit -> New -> S:\JDK\11\bin
```
Instead of `S:\JDK\11` use path to the folder where you unpacked JDK from https://jdk.java.net/archive/

I was able to build it and unlock bootloader (Latest Mi Unlocker though failed to do it...)

> ### PROBLEM SOLVED!
> The problem is that the version available for download on the official website and the release page is too old! The problem has been patched but you have to build it yourself [...]

Ernesto-Gallegos, 2022-01-27:
Hi, how I build in windows 10? I’m on desktop pc.

n0a, 2022-02-07:
**Build on M1 (must use i386 arch!!!):**

Download openjdk-11.0.2 (i386) first:

```
$ cd /Library/Java/JavaVirtualMachines/
$ sudo wget https://download.java.net/java/GA/jdk11/9/GPL/openjdk-11.0.2_osx-x64_bin.tar.gz
$ sudo tar xf openjdk-11.0.2_osx-x64_bin.tar.gz && sudo rm openjdk-11.0.2_osx-x64_bin.tar.gz
```

View Java versions: 
```
$ /usr/libexec/java_home -V
    11.0.2 (x86_64) "Oracle Corporation" - "OpenJDK 11.0.2" /Library/Java/JavaVirtualMachines/jdk-11.0.2.jdk/Contents/Home
```
Add $JAVA_HOME and choice OpenJDK 11.0.2:
```
$ nano ~/.zshenv
export JAVA_HOME=$(/usr/libexec/java_home -v11)

$ source ~/.zshenv
$ echo $JAVA_HOME
/Library/Java/JavaVirtualMachines/jdk-11.0.2.jdk/Contents/Home
```
Going to i386:
```
$ env /usr/bin/arch -x86_64 /bin/zsh --login
$ arch
i386
$ cd ~
$ git clone https://github.com/francescotescari/XiaoMiToolV2
$ cd XiaoMiToolV2
```

Add macOS fastboot and adb:
```
$ wget -O ./res/tools/adb https://github.com/francescotescari/XiaoMiToolV2/raw/mac/res/tools/adb && chmod +x ./res/tools/adb
$ wget -O ./res/tools/fastboot https://github.com/francescotescari/XiaoMiToolV2/raw/mac/res/tools/fastboot && chmod +x ./res/tools/fastboot
```
Build with run:
`$ ./gradlew build && ./gradlew run  `

**Work perfectly on MBP M1 with unlock Xiaomi Redmi 5 Plus bootloader without any cookie error.
Enjoy!**

dnarock666, 2022-03-23:
SOMEONE PLEASE THINK TO THE WINDOWS CHILDREN!
WTF SHOULD WE DO TO SOLVE THIS SHEEEIT?

Falcone8, 2022-03-29:
> **Build on M1 (must use i386 arch!!!):**
>
> Download openjdk-11.0.2 (i386) first:
>
> ```
> $ cd /Library/Java/JavaVirtualMachines/
> $ sudo wget [...]

On My MacBook Pro 13 M1 Doesn't work missing service token :(

alexbarcelo, 2022-04-29:
For any Linux users that stumble with this: before the `./gradlew run`, it seems to be necessary to fix the `fastboot` and `adb` binaries.

The easiest solution for me is to:
- Install them with your package manager (`sudo apt install adb fastboot` or equivalent)
- Add symbolic links to the `./res/tools` folder (`ln -s /usr/bin/adb ./res/tools/` and `ln -s /usr/bin/fastboot ./res/tools/`)

singhh9596, 2022-06-12:
can anyone share the updated binary my MacBook failing to compile?

HannesGitH, 2022-07-30:
well, you might as well give up, since even the correct version failes since #60