francescotescari/XiaoMiToolV2 #51: Can't login on Linux

MrOrangeMan, 2021-12-24:
It always says code is incorrect, even tho IT IS CORRECT, and i was able to login in browser

[image]

MariosK239, 2021-12-26:
I have forked the repository and applied all pull requests to the Linux branch, for ease of use here: https://github.com/MariosK239/XMT_V2-linux-fixed
You can git clone the repo and then follow the readme to compile and start the programme. It should work, as I used it today to unlock my bootloader.

MrOrangeMan, 2021-12-27:
> I have forked the repository and applied all pull requests to the Linux branch, for ease of use here: https://github.com/MariosK239/XMT_V2-linux-fixed You can git clone the repo and then follow the [...]

i was able to solve my issue using fastboot and adb old fashioned way xD
But thanks, this will be useful if someone else has this problem

Prodromo, 2022-03-03:
> I have forked the repository and applied all pull requests to the Linux branch, for ease of use here: https://github.com/MariosK239/XMT_V2-linux-fixed You can git clone the repo and then follow the [...]

I tried to use your version but it still doesn't show the chaptcha

[image]