francescotescari/XiaoMiToolV2 #53: Getting   FAILED (Command length to RawCommand() is too long)

dhvanilp, 2022-01-19:
[image]

dhvanilp, 2022-01-19:
I have fixed the url error using the branch from one of the PRs

dan4ik605743, 2022-02-03:
Same problem

dan4ik605743, 2022-02-03:
> I have fixed the url error using the branch from one of the PRs

How did you get it? Tell me more please.

Misterio77, 2022-04-21:
I got this fixed by using the developer provided adb/fastboot binaries instead of my distro's. This https://github.com/aosp-mirror/platform_system_core/blob/master/fastboot/fastboot_driver.cpp#L438 is probably patched somehow

RohitVermaOP, 2022-05-26:
> I got this fixed by using the developer provided adb/fastboot binaries instead of my distro's. This https://github.com/aosp-mirror/platform_system_core/blob/master/fastboot/fastboot_driver.cpp#L438 is probably patched somehow

How?