francescotescari/XiaoMiToolV2 #54: Redmi 9A bootloader unlock not working

eeskbala, 2022-02-09:
On a device that has developer mode enabled and Mii account registered, XiaoMiTool V2 fails to unlock the bootloader. It comes up with
Failed to get the device unlock token.

Error family code:info_retrive_failed
Error origin:com.xiaomitool.v2.procedure.install.FastbootInstall$4.run(Unknown Source)

Dav4555, 2022-04-10:
> On a device that has developer mode enabled and Mii account registered, XiaoMiTool V2 fails to unlock the bootloader. It comes up with Failed to get the device unlock token.
> 
> Error family code:info_retrive_failed Error origin:com.xiaomitool.v2.procedure.install.FastbootInstall$4.run(Unknown Source)

What OS u are trying? The Linux/Mac/Windows branches need fixes

k4r7hy, 2022-04-16:
i am suffering from the same problem i am using 
Linux archlinux 5.16.14-arch1-1 x86_64 GNU/Linux
and using latest version of tool

k4r7hy, 2022-04-22:
Hey if anyone looking to unlock a redmi 9a then google mediatek unlock

bbbenji, 2022-09-14:
> Hey if anyone looking to unlock a redmi 9a then google mediatek unlock

Thank you, that worked!

Nik-Kot, 2022-12-08:
I made a fix for that, not yet merged #88