francescotescari/XiaoMiToolV2 #56: Captcha image not showing

mushtaqx, 2022-02-27:
When i try to sign in, there is a captcha box but the app only shows an empty square.

I am on Linux but i doubt it would be any different for windows users?

[image]

Prodromo, 2022-03-03:
I have the exact same problem on Ubuntu. I noticed you can CTRL+A to copy and paste in a writer file and you would see the captcha. But when I type it's not accepted.
[image]

This way I cannot receive the text from Xiaomi.

Did you manage to solve it?

mushtaqx, 2022-03-05:
@Prodromo I noticed there was the option to login with qr code and that gives the exact same response as in your screenshot.

konorbi, 2022-03-06:
As a workaround I logged in with Facebook and there was no verification needed.

Dav4555, 2022-04-10:
> Captcha starts appearing if you bump the version of JavaFX, seems like an outdated WebView issue. I haven't noticed any GUI bugs in the login procedure (up to the SMS verification dialog) from doing this. Comment out the previous LTS and edit to newest build.gradle.kts, line 35:
> 
> `version = "11.0.2"`
> 
> to
> 
> `//version = "11.0.2"` `version = "17.0.2"`
> 
> Rerun `gradle build` and `gradle run` and it should work.

Thx this worked and I don't run in another in the other bug mentioned. I forked the tool with the fixes inside (url and captcha)

mushtaqx, 2022-04-15:
@Dav4555

Tried your fork, it seems Xiaomi is being a complete **** about it though...
[image]

Dav4555, 2022-04-16:
> @Dav4555
> 
> Tried your fork, it seems Xiaomi is being a complete **** about it though... [image]

In which step u run in that issue?

mushtaqx, 2022-04-16:
@Dav4555 
At the very last step, after giving permission and rebooting to fastboot.

Dav4555, 2022-04-26:
> @Dav4555 At the very last step, after giving permission and rebooting to fastboot.

Ok after waiting the amount of time I have the same Problem. I took a Windows notebook and used the official tool. Probably last Xiaomi phone

mushtaqx, 2022-05-04:
> > @Dav4555 At the very last step, after giving permission and rebooting to fastboot.
> 
> Ok after waiting the amount of time I have the same Problem. I took a Windows notebook and used the official tool. Probably last Xiaomi phone

I do understand that they don't want people to use their API through other frontends but at the same time i get a really bad vibe when they act nerdy meanwhile disregarding other OS users than Windows in order to unlock a freaking bootloader or perform other simple common tasks.

Not just not providing but also disallowing other frontends from providing those basic functionalities in a good faith.