francescotescari/XiaoMiToolV2 #57: Xiaomi procedure failed: [getServiceToken] Missing serviceToken cookie on windows

Kristalik0, 2022-02-27:
help please

p4kl0nc4t, 2022-04-17:
I have the same issue

Rafapadilla93, 2022-05-13:
To solve this error, you have to run the latest version by cloning the repository. I put a summary of how I have done it. I have Windows 11.

1- Install Visual Studio Code (for example).
2-Install Git: https://git-scm.com/
3-Install Java SE Development Kit: https://www.oracle.com/java/technologies/downloads/#jdk18-windows
4-Restart the computer
5-Open Visual Studio and Git Clone the repository 

> git clone https://github.com/francescotescari/XiaoMiToolV2

6-Once the repository is downloaded, you have to edit a file. Got to "Gradle/Wrapper/gradle-wrapper.propierties"
7-Modify the version line to:

> distributionUrl=https\://services.gradle.org/distributions/gradle-7.3.2-bin.zip

8-Go to View > Terminal
9-Execute _./gradlew build_
10-Execute _./gradlew run_

You can now use the tool.

Ivanturas, 2022-06-06:
Nope... didn't work for me... but tbh, it doesnt't seem like it's a XioaMiTool issue... seems to be a Xiaomi thing. They're messy af with their procedures, platforms and systems.

FFYzz, 2023-01-04:
after binding your xiaomi account and xiaomi device 168 hours, then you can use this tool to do something.

ka223, 2023-03-05:
Have the same problem on linux, the device is connected for 8 days now, how can I fix this issue?

aTm12, 2023-07-15:
guys i found the solution  finally, i  had this problem in  linux hen i jumped to windows same problem as well ... i remembered  that after  buying my first ciaomi  back in 2016  in order to  unlock ur phone u had to send a request to xiaomi in their website and they will send u a link to download mi unlock ... however  later they said  after u enable debugging on ur phone  u have to  turn off wifi  and then  register 4g  on ur sim  and then enter  developper options then  mi unlock status  and in there u have to press  add my account ..... and wait for the succesfully notiication .. in case u have an error  u have to go to settings and then mi account then sign out from ur acccount and login again and then enter developer options and mi unlock status and then press add my account  after u receive the succesfully notification u  can  unlock it in  the offical app (mi unlock tool)  i haven t tried  v2 app yet ...be careful don t change the sim number of ur xiaomi  account credentials otherwise u will have to wait 15 days to let u unlock it  
it's as  easy as this ....good luck

dharmtej-0206, 2023-11-08:
Where is it mentioned to wait 168 hrs

devrieshh, 2023-12-20:
dont forget this stap,in developer options you have to say no to oem unlock and usb debugging first,than restart,go bag to developer options 
find if you wish to unlock,say yes and reenter your xiaomi account(new account),after this enable usb debugging and oem unlock.
then start  mi unlock tool