francescotescari/XiaoMiToolV2 #58: A way to transfer authentication data from official tool or browser to XMT2?

beefmode, 2022-03-03:
This has to do with unlocking the bootloader on my Xiaomi phone.

While solving the issue of the invisible CAPTCHA ([https://github.com/francescotescari/XiaoMiToolV2/issues/56](url)), I've run into another one. The SMS code required to login using the new URL (patched line 39) never arrives. This only happens in XMT2, both the official tool and my browser (Firefox latest) work just fine using the same link formed with a `locale=en`.

Where and how does XMT2 store the login credentials I need to unlock my phone? I'd like to manually bypass the authentication process and just supply everything necessary to the adb & fastboot steps, if there's such a way.

mushtaqx, 2022-03-05:
I did some digging and it turns out that XMT2, simply opens a browser for login and then after login it just takes the cookies from browser to proceed with the rest of the operation.
Its stored in memory unless you specify to store it in a file.

konorbi, 2022-03-06:
As a workaround I logged in with Facebook and there was no verification needed.

anbuchelva, 2022-04-12:
Try using the QR code method to login. I have the same issue but it works.