francescotescari/XiaoMiToolV2 #61: Failed to get security code [ArchLinux]

skylabhalo, 2022-03-27:


dumbPy, 2022-03-27:
[image]
I am stuck at this point, where the captcha always fails...  
Logging in from the UI, it doesn't uses this type of captcha, but instead uses google's reCaptcha

I guess it's something recent change that someone might have to figureout before the tool starts working again

surender77180, 2022-03-28:
clone the repo first 
install java and javafx
compile using gradlew build
run using gradlew run

skylabhalo, 2022-03-28:
build fails

edisionnano, 2022-04-13:
> build fails
> 
> ```
>  Task :compileJava FAILED
> 
> FAILURE: Build failed with an exception.
> 
> * What went wrong:
> Execution failed for task ':compileJava'.
>  java.lang.IllegalAccessError: class org.gradle.internal.compiler.java.ClassNameCollector (in unnamed module @0x6b4a0933) cannot access class com.sun.tools.javac.code.Symbol$TypeSymbol (in module jdk.compiler) because module jdk.compiler does not export com.sun.tools.javac.code to unnamed module @0x6b4a0933
> 
> * Try:
> Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
> 
> * Get more help at https://help.gradle.org
> ```
sudo pacman -Syu jdk11-openjdk && sudo archlinux-java set java-11-openjdk

rafniedz, 2022-05-04:
I'm experiencing getServiceToken issue on LinuxMint for over a month. Is there any solution for this?

[skylabhalo](https://github.com/skylabhalo) is the solution 
"sudo pacman -Syu jdk11-openjdk && sudo archlinux-java set java-11-openjdk"
successful ?

rafniedz, 2022-05-04:
I've tested the same scenario on Windows version (Win10, current updates) and the result is the same. It seems that error is platform-independent...
[image]

edisionnano, 2022-05-10:
> I'm experiencing getServiceToken issue on LinuxMint for over a month. Is there any solution for this?
> 
> [skylabhalo](https://github.com/skylabhalo) is the solution "sudo pacman -Syu jdk11-openjdk && sudo archlinux-java set java-11-openjdk" successful ?

My "solution" is just for building the app, it won't solve the tokken issue

ClockwiseLabs, 2022-06-16:
I think i fount the solution!
I spent 3 day trying to unlock the bootloader of my redmi 4x, without success.
The token error was do to my miui account not matching the account with wich i was logged in on my phone.
Eaven if you email and password match, Xiaomi can create two different account ID's. The double ID problem was do to the possibilty to recive the sms confirmation with the same phone number, with or without "0" at the beginning. Like "+44076xxxx190" or "+4476xxxx190".
The second error was the fastboot connection error. This is do to the screen lock.
You need to turn off all screen lock password, fingerprints and patterns. Also, VERY IMPORTANT, on the developer option menu you need to to turn on "skip screen lock" and "stay awake option". The phone needs to be active all the time eaven during the fastboot screen.
Reply if you are successful!

Blocksrey, 2022-06-22:
Does this mean you should or shouldn't include the preceding 0? Please let me know :)

samyak039, 2023-11-27:
> I think i fount the solution! I spent 3 day trying to unlock the bootloader of my redmi 4x, without success. The token error was do to my miui account not matching the account with wich i was logged [...]

Sadly, this method didn't work for me :(