francescotescari/XiaoMiToolV2 #64: Fails to save login info

davispuh, 2022-04-01:
Even when using latest git version, it still fails to save login info.

Basically login flow works correctly. I enter user/pass/captcha then SMS code and it shows me
```
{"R":"","S":"OK"}
```

[image]

I see that if you open in browser https://account.xiaomi.com/pass/serviceLogin?sid=unlockApi&json=false&passive=true&hidden=false&_snsDefault=facebook&checkSafePhone=true&_locale=en
it gives cookies and automatically redirects to `https://unlock.update.miui.com/sts`

I guess our browser instance gets redirected without having a chance to grab cookies.

Dav4555, 2022-04-10:
> Even when using latest git version, it still fails to save login info.
>
> Basically login flow works correctly. I enter user/pass/captcha then SMS code and it shows me
>
> ```
> {"R":"","S":"OK"}
> ``` [...]

On which OS? Which latest git version u are using?I don't have that problem.

davispuh, 2022-04-10:
On ArchLinux, using a335da6bb812272d68d6842ff88c4b7552465cc4 but on Linux and I applied Linux commit on top.

Dav4555, 2022-04-10:
> On ArchLinux, using [a335da6](https://github.com/francescotescari/XiaoMiToolV2/commit/a335da6bb812272d68d6842ff88c4b7552465cc4) but I applied Linux commit on top.

Try my fork Linux branch, I use Manjaro.

Dav4555, 2022-04-12:
> > > On ArchLinux, using [a335da6](https://github.com/francescotescari/XiaoMiToolV2/commit/a335da6bb812272d68d6842ff88c4b7552465cc4) but I applied Linux commit on top.
> > 
> > 
> > Try my fork Linux branch, I use Manjaro.
> 
> It works only for the SMS based auth; not the QR code based auth. The problem is that the SMS code arrives once out of 5+ tries.

u only need to login once, go to settings and save the login data. then push the login. dont login while the unlock process.

m911t, 2022-07-30:
your fix doesnt work either.

immjs, 2022-12-23:
> It works only for the SMS based auth; not the QR code based auth. The problem is that the SMS code arrives once out of 5+ tries.

Further in post (also my case)
> I enter user/pass/captcha then **SMS code** and it shows me