francescotescari/XiaoMiToolV2 #65: Seems MIUI API have changed

davispuh, 2022-04-01:
I have `Xiaomi 11T Pro` (codename `vili`) and most of APIs failed to work.

```
Looking for latest recovery rom version using apis
POST https://update.miui.com/updates/v1/latestverinfo.php
sid=miassistant&d=vili_global&c=12&f=1&r=XXX&s=YYY
OTA Recovery response: {"code":0,"data":"param error"}
```

```
Basic ota request found 0 valid roms
InstallException created: missing_property - Ota response doesn't contain an installable rom data
Fallback: skipping install exception: missing_property - Ota response doesn't contain an installable rom data
InstallException - rom_selection_error - Cannot find any rom that can recover your device
```

```
POST https://update.miui.com/updates/v1/devinfo.php
sid=miassistant&t=1001&r=XXX&s=YYY
```
This phone is not in this list at all.

```
https://update.miui.com/updates/v1/fullromdownload.php?d=vili&b=F&r=cn&n=
http://update.miui.com/updates/miota-fullrom.php?d=vili&b=X&r=cn&n=&l=en_US
https://update.miui.com/updates/v1/fullromdownload.php?d=vili&b=X&r=cn&n=
https://update.miui.com/updates/v1/fullromdownload.php?d=vili_global&b=X&r=global&n=

```
all of these redirected to http://www.miui.com/

But these seem to work:
* `http://update.miui.com/updates/miota-fullrom.php?d=vili_global&b=F&r=global&n=&l=en_US`
* `http://update.miui.com/updates/miota-fullrom.php?d=vili_eea_global&b=F&r=eea&n=&l=en_US`
* and encrypted `https://update.intl.miui.com/updates/miotaV3.php` (has newer ROMs than others)

PS. I'm using latest git version

davispuh, 2022-04-10:
I found that this works https://update.miui.com/updates/v1/fullromdownload.php?d=vili_global&b=F&r=global&n=  (notice `b=X` replaced with `b=F`) if you pass `Referer: https://c.mi.com/`

davispuh, 2022-04-12:
```
https://update.miui.com/updates/v1/fullromdownload.php?d=vili&b=F&r=cn&n=
http://update.miui.com/updates/miota-fullrom.php?d=vili&b=X&r=cn&n=&l=en_US
https://update.miui.com/updates/v1/fullromdownload.php?d=vili&b=X&r=cn&n=
https://update.miui.com/updates/v1/fullromdownload.php?d=vili_global&b=X&r=global&n=
```

These ^ failing are not an issue, XiaoMiToolV2 tries to download all roms by trying different regions with different branches (stable, dev) and it's normal that only some will work for some devices.

This one `https://update.miui.com/updates/v1/latestverinfo.php` is actually broken currently but I don't think it's that useful because `miota-fullrom.php` (for full fastboot roms) and `miotaV3.php` (for OTA sideloading) works. But I noticed that even these they don't return latest ROM info :c don't know how https://github.com/XiaomiFirmwareUpdater/miui-updates-tracker are doing it but they have latest firmware on https://xiaomifirmwareupdater.com/ where it's newer than all the sites and all the links I saw anywhere else.

koyuyesil, 2022-07-07:
hello recovery flash api how to work im stuck at here. i want use command line.

eErenYeager, 2022-09-06:
I have found a way to flash recovery rom using xiamitool V2.

koyuyesil, 2022-09-25:
exit miflash pro

adb.exe kill-server
recovery.exe getinfo

get api parameters from miassitant sideload mode

qianmo-xw, 2022-10-03:
> I have found a way to flash recovery rom using xiaomitool V2. By changing the region and device in code

I can not understand it，can you give me more information？

eErenYeager, 2022-10-03:
> > I have found a way to flash recovery rom using xiaomitool V2. By changing the region and device in code
> 
> I can not understand it，can you give me more information？

When I connected my device in recovery mode. This tool was not able to detect my device name and region. Also my device name is missing in the list of devices this program provides. So I added my device name manually in this line
https://github.com/francescotescari/XiaoMiToolV2/blob/a335da6bb812272d68d6842ff88c4b7552465cc4/src/main/java/com/xiaomitool/v2/xiaomi/XiaomiUtilities.java#L359

After that I have to change the region to correct region then only it started downloading the device ota and started flashing.
The default region it use is global. But my region was india so I changed the region to India by replacing "GLOBAL_STABLE" to "INDIA_STABLE" in this line:
https://github.com/francescotescari/XiaoMiToolV2/blob/a335da6bb812272d68d6842ff88c4b7552465cc4/src/main/java/com/xiaomitool/v2/procedure/fetch/StockRecoveryFetch.java#L367

The list of possible region name can be found here:
https://github.com/francescotescari/XiaoMiToolV2/blob/a335da6bb812272d68d6842ff88c4b7552465cc4/src/main/java/com/xiaomitool/v2/rom/MiuiRom.java#L126