francescotescari/XiaoMiToolV2 #68: Xiaomi sideload

davispuh, 2022-04-11:
Does anyone know anything about how sideloading works for Xiaomi deices?
How that protocol is like? Seems every manufacturer customizes it and standard `adb sideload` doesn't work.

I have Xiaomi 11T Pro and I've been trying to sideload OTA image but it just doesn't work.
When you boot recovery there is option with "Connect with MIAssistant"
I tried using  `MiPhoneAssistant (4.2 Beta)` but it doesn't work, after logging in, webpage just stays.
Then I tried with `adb sideload` from it's folder but it just gives
```
connecting...
* failed to read command: No error
```

and `adb` from this project
```
[0/3188823602]          adb: failed to read command: No error
```

this OTA update zip file contains
```
META-INF/com/android/metadata
META-INF/com/android/metadata.pb
META-INF/com/android/otacert
apex_info.pb
care_map.pb
payload.bin
payload_properties.txt
```

`fastboot update` fails with
```
--------------------------------------------
Bootloader Version...: 
Baseband Version.....: 
Serial Number........: 
--------------------------------------------
archive does not contain 'android-info.txt'
fastboot: error: could not read android-info.txt
```

I just found [XiaomiADB](https://www.xiaomitool.com/latestadb) but even that fails (also seems sadly it's not open source)
```
> XiaomiADB.exe sideload_miui miui_VILIEEAGlobal_V13.0.8.0.SKDEUXM_5142aa18f5_12.0.zip
* Miui sideload made by Francesco Tescari, updated 02 Feb 2018 *
* Checking device status *
* Getting recovery info *
* Creating file MD5 hash *
* Querying ota update server, attempt #0 *
ERROR: This rom cannot be installed, server code -> 2015, server message -> Couldn't verify update package. Update via USB instead.
* Querying ota update server, attempt #1 *
ERROR: This rom cannot be installed, server code -> 2015, server message -> Couldn't verify update package. Update via USB instead.
* Querying ota update server, attempt #2 *
ERROR: This rom cannot be installed, server code -> 2015, server message -> Couldn't verify update package. Update via USB instead.
* Querying ota update server, attempt #3 *
ERROR: This rom cannot be installed, server code -> 2015, server message -> Couldn't verify update package. Update via USB instead.
ERROR: Installation of this rom via stock recovery is not allowed by Xiaomi
```

Then finally it should be possible to update by placing zip OTA file inside `/sdcard/downloaded_rom`  and then updating from settings (as per https://c.mi.com/oc/miuidownload/detail?guide=1). But this MIUI version 13.0.2 doesn't even have such option there... WTF...

davispuh, 2022-04-11:
[image]

davispuh, 2022-04-12:
Okay lol so it turns out I had to click couple of times on "13" logo for this option to get enabled and show up.

But then when I tried to update it still failed with error "Can't verify update"
[image]

Only later I realized that's because I was trying to apply EEA OTA onto global build. But error message really could have been more clear... It worked fine after changing OTA file. And path where you place it doesn't matter at all, but need to go by full path, it doesn't work by shortcuts such as "recent" places.

But none of adb nor XiaomiADB still doesn't work even with correct OTA file. So seems only is to update from settings app which isn't that nice.

vladimir-kuznetsov, 2024-02-15:
ADB sideload still works but you need a proper binary:
https://www.xiaomitool.com/adb

chalakjisu161-del, 2026-10-06:
### ### ### ### ### _**jjjjuuuuuuu**_