francescotescari/XiaoMiToolV2 #70: Redmi Note Pro 6 not recognize ?

Jacques-Olivier-Farcy, 2022-05-16:
Hi,

I am stuck with the official tool at this point with my model M1896E7TG
[image]

I tried your tool, but nothing happened, maybe i ma not doing the right thing ?

Actualy i tried with my phone with USB enabled:

1. Boot to the fastboot with vol down + Powr key
2. When the logo of fastboot arrived , realese the keys
3. Launch XiaoMiToolV2
4. Connect usb phone to PC (Windows 10)

Jacques-Olivier-Farcy, 2022-05-16:
I had this on screen:
[image]

But when i try to select i have an error telling me i have to enable USB debugging, but i am in fastboot mode.
…

Jacques-Olivier-Farcy, 2022-05-16:
I sent feedback from the tool

nocturn9x, 2022-06-04:
This has been already addressed in the other issue and has been fixed for ages. Check the issue conversation and apply the patch.