francescotescari/XiaoMiToolV2 #73: Invalid fastboot token. Please retry or upgrade your unlock tool.

StupidRepo, 2022-06-10:
I keep getting this error when rooting my Xiaomi Redmi Note 10 5G (I got it 4 days ago and linked it to my Mi account 3 days ago).

I cloned from git and built it aswell.

singhh9596, 2022-06-12:
same. tried many versions on mac. most of them were crasing and some of them couldn't be opened. if by any luck they ran, while trying to unlocking my device I'm running into the same error code.

ClockwiseLabs, 2022-06-16:
I think i fount the solution!
I spent 3 day trying to unlock the bootloader of my redmi 4x, without success.
The token error was do to my miui account not matching the account with wich i was logged in on my phone. 
Eaven if you email and password match, Xiaomi can create two different account ID's. The double ID problem was do to the possibilty to recive the sms confirmation with the same phone number, with or without "0" at the beginning. Like "+44076xxxx190" or "+4476xxxx190".
The second error was the fastboot connection error. This is do to the screen lock. 
You need to turn off all screen lock password, fingerprints and patterns. Also, VERY IMPORTANT, on the developer option menu you need to to turn on "skip screen lock" and "stay awake option". The phone needs to be active all the time eaven during the fastboot screen.
Reply if you are successful!