francescotescari/XiaoMiToolV2 #76: Is there an alternative to XiaoMitoolV2? Yes, miunlock

Attigliuzzo, 2022-08-10:
The [tool](https://github.com/Canny1913/miunlock) is written in Python, so multi-platform.
I doesn't test a full unlock of my Poco because it is still under warranty but i tested until a point.
An user on XDA goes [a bit forward](https://forum.xda-developers.com/t/tool-win-lin-mac-miunlocktool-unlock-bootloader-of-xiaomi-devices-on-mac-linux.3782444/post-87265441).
Seems that it does its job!!!

ViBE-HU, 2022-08-10:
unlocking is not a dedicated feature of XiaoMiTool. even XiaoMiTool is not an unlocker. so **miunlock** cannot be an alternative. if you want to unlock on Linux it's another scenario since this XiaoMiTool use the official unlocking tool and it has no Linux version or any Linux binaries.

unfairDude, 2022-08-12:
patched

Practicalbutterfly5, 2022-08-13:
> patched

Is this tool patched as well?

unfairDude, 2022-08-13:
> > patched
> 
> Is this tool patched as well?

Yes