francescotescari/XiaoMiToolV2 #79: Error 20045: Please use the common user tool on the official website

Andre8900, 2022-09-09:
Hello, I want to unlock my bootloader, but I get the following response (in Linux) after confirming all dialogs:

Unlock request response: {"code":20045,"description":"请使用官网普通用户解锁工具进行操作","descCN":"请使用官网普通用户解锁工具进行操作","descEN":"Please use the common user tool on the official website.","uid":"<retracted>"}

Andre8900, 2022-09-09:
Phone is Poco F3 (alioth) and I am using the latest version with the sessionToken fix from #23 (https://github.com/francescotescari/XiaoMiToolV2/issues/23#issuecomment-904082515)

LampPrinter, 2022-10-11:
Same error.

I was finally able to get past "service token error" by following this thread.
https://forum.xda-developers.com/t/tool-win-lin-mac-miunlocktool-unlock-bootloader-of-xiaomi-devices-on-mac-linux.3782444/page-13#post-85932539

Now I am getting "Failed to unlock your device, Xiaomi server returned error 20045"

shubham-cpp, 2022-11-10:
Is this project dead?

Nik-Kot, 2022-12-08:
Download the latest version using `git clone https://github.com/francescotescari/XiaoMiToolV2.git` and run it with `gradlew run`. This issue is already fixed.

pedrofsn, 2022-12-27:
@Nik-Kot I've cloned the project, build and run. But with this the software can't found some important pieces and was not working as well (with fonts).

> Cannot execute fastboot command "fastboot devices", reason: Cannot run program "./res/tools/fastboot": error=2, No such file or directory

Any tips?
[image]

pedrofsn, 2022-12-27:
I'm runnig MacOS Monterrey. I checked the folder and just found `.exe` files. I need to specify some argument when I run `.gradlew`?

pedrofsn, 2022-12-27:
I've changed to `mac` git branch and now I've right files on `tools` folder. 
I don't know how to fix Font errors yet but app is running `fastboot` as well.

pedrofsn, 2022-12-27:
[image]

Looking to the terminal, I'm getting the same error from issues: #85 #61 etc

```
[10:28:05][WARN  ][52a4d629] Pre-unlock requests failed [getServiceToken] Missing serviceToken cookie
[10:28:05][ERROR ][52a4d629] com.xiaomitool.v2.xiaomi.XiaomiProcedureException: [getServiceToken] Missing serviceToken cookie
        at com.xiaomitool.v2.xiaomi.XiaomiServiceEntry.httpGetServiceToken(XiaomiServiceEntry.java:89)
        at com.xiaomitool.v2.xiaomi.XiaomiServiceEntry.requireSSandST(XiaomiServiceEntry.java:129)
        at com.xiaomitool.v2.xiaomi.XiaomiKeystore.requireServiceKeyAndToken(XiaomiKeystore.java:69)
        at com.xiaomitool.v2.xiaomi.unlock.UnlockRequest.exec(UnlockRequest.java:37)
        at com.xiaomitool.v2.xiaomi.unlock.UnlockCommonRequests.nonceV2(UnlockCommonRequests.java:100)
        at com.xiaomitool.v2.xiaomi.unlock.UnlockRequest.addNonce(UnlockRequest.java:80)
        at com.xiaomitool.v2.xiaomi.unlock.UnlockCommonRequests.userInfo(UnlockCommonRequests.java:116)
        at com.xiaomitool.v2.procedure.install.FastbootInstall$4.run(FastbootInstall.java:250)
        at com.xiaomitool.v2.procedure.RInstall.runInternal(RInstall.java:23)
        at com.xiaomitool.v2.procedure.RNode$RNodeSequence.run(RNode.java:108)
        at com.xiaomitool.v2.procedure.RNode.runInternal(RNode.java:77)
        at com.xiaomitool.v2.procedure.RNode$RNodeSequence.run(RNode.java:108)
        at com.xiaomitool.v2.procedure.RNode.runInternal(RNode.java:77)
        at com.xiaomitool.v2.procedure.RNode$RNodeSequence.run(RNode.java:108)
        at com.xiaomitool.v2.procedure.Procedures$6.run(Procedures.java:236)
        at com.xiaomitool.v2.procedure.RInstall.runInternal(RInstall.java:23)
        at com.xiaomitool.v2.procedure.RNode$RNodeSequence.run(RNode.java:108)
        at com.xiaomitool.v2.procedure.RNode.runInternal(RNode.java:77)
        at com.xiaomitool.v2.procedure.RNode$RNodeSequence.run(RNode.java:108)
        at com.xiaomitool.v2.procedure.RNode.runInternal(RNode.java:77)
        at com.xiaomitool.v2.procedure.RNode$RNodeSequence.run(RNode.java:108)
        at com.xiaomitool.v2.engine.actions.ActionsDynamic$5.run(ActionsDynamic.java:604)
        at com.xiaomitool.v2.procedure.install.GenericInstall$10$1.run(GenericInstall.java:292)
        at java.base/java.lang.Thread.run(Thread.java:829)

```

pedrofsn, 2022-12-27:
I've tried with @edd34 `fix/mac_login_url` branch (https://github.com/edd34/XiaoMiToolV2/tree/fix/mac_login_url) but still getting same error.
Looking for request https://unlock.update.miui.com/sts and getting 401 Unauthorized.

drendog, 2022-12-28:
> Hello, I want to unlock my bootloader, but I get the following response (in Linux) after confirming all dialogs:
> 
> Unlock request response: {"code":20045,"description":"请使用官网普通用户解锁工具进行操作","descCN":"请使用官网普通用户解锁工具进行操作","descEN":"Please use the common user tool on the official website.","uid":""}

I fixed it by creating another mi account with same number phone

Neclude, 2023-01-13:
> I fixed it by creating another mi account with same number phone

How you do it? Are you waited 7 days again with new mi account?

Nik-Kot, 2023-01-15:
> I've changed to `mac` git branch and now I've right files on `tools` folder. I don't know how to fix Font errors yet but app is running `fastboot` as well.

Sorry, the `mac` branch was notpatched at the moment, only the master branch was. Now I'm working on it and also on releasing a compiled binary.

drendog, 2023-01-28:
> How you do it? Are you waited 7 days again with new mi account?

yes

xintrea, 2023-08-31:
Same trouble at Debian Linux 11 Stable & POCO M4 Pro

salhgmal, 2024-04-15:
> قم بتنزيل أحدث إصدار باستخدامه `git clone https://github.com/francescotescari/XiaoMiToolV2.git`وقم بتشغيله باستخدام `gradlew run`. تم إصلاح هذه المشكلة بالفعل.

Whether the problem is solved or not xiaomi server returned error 20045

salhgmal, 2024-04-15:
xiaomi server reurned error 20045

xintrea, 2024-04-26:
> xiaomi server reurned error 20045
And what should I do now?