francescotescari/XiaoMiToolV2 #85: selene oem unlock getServiceToken missing cookie

CastixGitHub, 2022-10-26:
Device: Redmi 10 selene

more than 15 days since device->account connection

checked the account ID is the same (by mistake i previously logged into an old one)

unable to unlock

tool version: V20.7.28(beta)

(unable to copy the traceback since the text on screen is not selectable and no logs on stderr)

Nik-Kot, 2022-12-08:
Download the latest version using `git clone https://github.com/francescotescari/XiaoMiToolV2.git` and run it with `gradlew run`. This issue is already fixed.

paolomeraviglia, 2022-12-15:
[image]

tool version: latest dev branch, linux

rosembergdesantana, 2023-02-13:
Still not working, logging the code the response get in this method: httpGetServiceToken() in this class XiaomiServiceEntry.

        EasyResponse response;
        Log.log_private("Second step request: " + url);
        response = EasyHttp.get(url); -> 401 response code.
        HashMap<String, String> cookies = response.getCookies();
        serviceToken = cookies.get("serviceToken");

Always getting 401 after the request trying to get the url base mount in this method signedLocation().