francescotescari/XiaoMiToolV2 #93: Xiaomitool V2 can't find ROM

RSK735, 2023-02-01:
When I'm on the step of going into the option of "Connect with MIAssistant" I check that no error of "MIUI version can't be installed in this device" popped in the phone screen, after I click "no" the program can't find no ROM and immediately gives me this code error:

[image]

This happens in version V20.7.21 and V20.7.28, don't matter which ROM I choose, the result is the same.
Firewall is not an issue, app has no restrictions and can connect to both networks.
Any help  will be appreciated!

maxbolgarin, 2023-02-01:
Same actions, same issue

[image]

Xiaomi mi 9 lite

vipmaster23, 2024-01-15:
+