francescotescari/XiaoMiToolV2 #104: Xiaomi procedure failed: [getServiceToken] Missing serviceToken cookie

kambei, 2023-05-17:
as #23 

[image]

on Linux.

kambei, 2023-05-17:
Following [this HOW-TO](https://cynici.github.io/2021/08/29/unlock-poco-f1-boot-loader-using-linux.html)

[image]

dingodoppelt, 2023-05-23:
same issue

StitiFatah, 2023-06-25:
Same, do you guys found anyway to unlock our device on Linux ? With the official app + bottle my phone doesn't get recognized

dkoek, 2023-07-09:
same problem on topaz

n1lby73, 2023-07-15:
same error here, any fix yet

Theend21, 2023-09-23:
fix found

[image]

StitiFatah, 2023-09-23:
> fix found
> 
> [image]
> 

And what is it ?

Belkachait08, 2023-09-27:
> fix found
> 
> [image]
Could you please share the fix?

Mykola-Veryha, 2023-10-01:
Unlocked it with Virtualbox and Windows 10

- load the phone into the `fastboot` mode
- start the Windows virtual machine
- connect the USB so windows can see the phone as Android device
- install USB driver (https://developer.android.com/studio/run/win-usb)
- download and install Mi flash unlock (http://miuirom.xiaomi.com/rom/u1106245679/7.6.727.43/miflash_unlock_en_7.6.727.43.zip)
- start Mi flash unlock and unlock the phone

Windows 10 ISO
https://www.microsoft.com/en-us/software-download/windows10ISO

USB driver 
https://developer.android.com/studio/run/win-usb

Mi flash unlock
http://miuirom.xiaomi.com/rom/u1106245679/7.6.727.43/miflash_unlock_en_7.6.727.43.zip

rxmmah, 2023-10-16:
> Unlocked it with Virtualbox and Windows 10
>
>     * load the phone into the `fastboot` mode
>
>     * start the Windows virtual machine
>
>     * connect the USB so windows can see the phone as Android [...]

how to install usb drivers, i got only files

n1lby73, 2023-12-17:
> Unlocked it with Virtualbox and Windows 10
>
>     * load the phone into the `fastboot` mode
>
>     * start the Windows virtual machine
>
>     * connect the USB so windows can see the phone as Android [...]

i wouldn't really say this is a fix to the issue instead it's a workaround to it as what you've done is simply running windows in a virtual machine and installing the authentic mi unlock tool that is only windows compatible

shauryasonar, 2023-12-25:
[image]
any fix?

shauryasonar, 2023-12-25:
> ![Screenshot 2023-12-25 123401](https://private-user-images.githubusercontent.com/106643251/292724343-ceb0bfa2-7edb-41fa-9857-0e0c442b5214.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM0ODgxOTIsIm5iZiI6MTcwMzQ4Nzg5MiwicGF0aCI6Ii8xMDY2NDMyNTEvMjkyNzI0MzQzLWNlYjBiZmEyLTdlZGItNDFmYS05ODU3LTBlMGM0NDJiNTIxNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIyNVQwNzA0NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04MWQxZGY1ZDIyMGJiNmJlMDE3YWUzY2I1NjBlYTBkYTYzMzM1OTkzNGI3NGVkOWY2YmE4YWIxNmJlNTlmNDBiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.4EW4uK7DmHGnjsaO1T1FQ9Wbu5ykVmQ-Wcz79weUuZE) any fix?

i m using win 10

rfc-2549, 2024-01-17:
> fix found
> 
> ![Screenshot 2023-09-23 194125](https://private-user-images.githubusercontent.com/145863560/270123455-ec54e99b-38d6-468f-bbc9-dc275d199538.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU1MDE4ODMsIm5iZiI6MTcwNTUwMTU4MywicGF0aCI6Ii8xNDU4NjM1NjAvMjcwMTIzNDU1LWVjNTRlOTliLTM4ZDYtNDY4Zi1iYmM5LWRjMjc1ZDE5OTUzOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExN1QxNDI2MjNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03Yjg5NTFmZTk2ZjZiYjY4ZmYzMjhkMjkxYTVlMDcxZTcwMmQzMzQyODk3YzI4ODczNmUyYjcxMjhiMTJiNmU3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.vGS_c_CeTwGHqFgwe5dmGrNZ0htAeuvcJP4595IpHVI)

Your life is worthless and you serve 0 purpose.

mindtheegab, 2024-01-31:
same issue on poco f5

carbon-starlight, 2024-12-22:
> fix found
>
> ![Screenshot 2023-09-23 [...]

What's the fix?