francescotescari/XiaoMiToolV2 #105: i cant unlock my phone bootloader

trung789, 2023-05-28:
screenshot I took
[image]

tgvqwsjk4jcs, 2023-07-27:
[image]
same

dongerlel, 2023-08-08:
why exactly would you use third party tool instead the official one from xiaomi???

muqiuhan, 2023-08-09:
> why exactly would you use third party tool instead the official one from xiaomi???

Because Xiaomi official tool does not support linux, we are linux or mac users

muqiuhan, 2023-08-09:
[image]

same

5ulo, 2023-08-10:
exactly the same here.. I have access only to macs m1 and linux devices. Tried Win 11 under UTM (mac) - adb is working fine, fastboot does not see devices. Tried in Virtualbox (linux) with Win 10 - the usb passthrough wont let the guest to see the phone. So that's why I used this tool which sadly does not work for me.

flixcrls, 2023-08-15:
same, wine doesnt show sign in, UTM and pararel desktop cant detect device and doesnt unlock

vsvsav, 2023-10-09:
> > why exactly would you use third party tool instead the official one from xiaomi???
> 
> Because Xiaomi official tool does not support linux, we are linux or mac users

It does not work for Win10 too.

t2hv33, 2023-10-11:
OK, so it's not working right? I think I need to borrow the Windows machine or open the old one. I currently have a Mac M1, but I think there's no luck here @trung789 @flixcrls @vsvsav @5ulo

flixcrls, 2023-10-11:
> OK, so it's not working right? I think I need to borrow the Windows machine or open the old one. I currently have a Mac M1, but I think there's no luck here @trung789 @flixcrls @vsvsav @5ulo

its not working for now, if u have windows pc go for it...

vsvsav, 2023-10-13:
> > OK, so it's not working right? I think I need to borrow the Windows machine or open the old one. I currently have a Mac M1, but I think there's no luck here @trung789 @flixcrls @vsvsav @5ulo
> 
> its not working for now, if u have windows pc go for it...

For desktop PC Win10 it is not working.

flixcrls, 2023-10-13:
@vsvsav thats hilarious, the windows cant ubl too? so for what that shitty company tools come in at first place, for protect user? or too add more waste?, have you try another version of the tools? if you already try, thats sucks 🤮

vsvsav, 2023-10-13:
> @vsvsav thats hilarious, the windows cant ubl too? so for what that shitty company tools come in at first place, for protect user? or too add more waste?, have you try another version of the tools? if you already try, thats sucks 🤮

Yeah, I tried a lot of. Bricked device will go to trash dump.

I tried:
MiFlashPro 7.3.706.21, batch_bl_to_edl 6.5.817.32, MiFlash 2022.5.7.0, Mi Unlocker 7.6.727.43
I tried this all:
[image]
Old tools could not to login, no codes to SMS or Viber.

Tried all this:
[image]

That is interesting, that these:
MiPhone20151028 and XiaoMiFlash(Beta)  (beta!! - unique version) 
could write with without errors (big log file) , but as I understand the problem that the locked partition only has a corrupted
data and could not be rewritten, these tools maybe ignored write errors.

flixcrls, 2023-10-13:
> > @vsvsav thats hilarious, the windows cant ubl too? so for what that shitty company tools come in at first place, for protect user? or too add more waste?, have you try another version of the tools? if you already try, thats sucks 🤮
> 
> Yeah, I tried a lot of. Bricked devices will go to trash dump.
> 
> I tried: MiFlashPro 7.3.706.21, batch_bl_to_edl 6.5.817.32, MiFlash 2022.5.7.0, Mi Unlocker 7.6.727.43 I tried this all: [image] Old tools could not to login, no codes to SMS or Viber (they changed sms provider for new version).
> 
> Tried all this: [image]
> 
> That is interesting, that these: **MiPhone20151028** and **XiaoMiFlash(Beta)** (beta!! - unique version) were able to write with without errors (large log file) , but AFAIK the problem that the locked partition only has a corrupted data and cannot be overwritten, these tools can ignore write errors or not everything is rewritten, something still remains corrupted. The latest versions of the utilities say right away, the partition is locked, go far.

about damn time that company shouldnt user ubl anymore, and i wish the company very bankrupt, remember that company get attention with the affordable device and easy to modding, the ubl came out in public since 2016, the company said its to protect the user(my ass), see the phone get more spyware and bloatware if you dont delete it and thats make your phone laggy day by day

flixcrls, 2023-10-13:
is it possible to remove the lock unlock bootloader system? im curious about that system 🤔

Dav4555, 2023-11-13:
> exactly the same here.. I have access only to macs m1 and linux devices. Tried Win 11 under UTM (mac) - adb is working fine, fastboot does not see devices. Tried in Virtualbox (linux) with Win 10 - [...]

I found a solution. U need to add your user to the group vboxusers. After that I could add a USB filter for my device in virtualbox.