francescotescari/XiaoMiToolV2 #106: Xiaomi procedure failed: [getServiceToken] Missing serviceToken cookie

KiraiEEE, 2023-05-28:
[image]

hicaru, 2023-06-05:
same problem paco f1

ztx-lyghters, 2023-06-20:
same problem on POCO M4 5G

closed 2023-06-22 by KiraiEEE

5ulo, 2023-08-10:
why is this closed? No solution shown.

Uhter, 2024-06-29:
> why is this closed? No solution shown.

It's a question I ask myself every time I come to github.

rizwan1239, 2024-08-19:
> > why is this closed? No solution shown.
> 
> It's a question I ask myself every time I come to github.

real

Milk-Cool, 2024-09-13:
@5ulo because it's a duplicate issue

Uhter, 2024-09-13:
> @5ulo because it's a duplicate issue

How can we know that?

Milk-Cool, 2024-09-13:
@Uhter #23

Uhter, 2024-09-13:
> @Uhter #23

Ok but… here, nothing tells us why it's closed.   
That's the point.

Deb-svg, 2024-09-18:
I have a solution

Deb-svg, 2024-09-18:
open the tool github page, then you go see the src/main/java/com/xiaomitool/v2 now you get to the XiaomiServiceEntry.java, then find the URL_FIRST inside the file code and copy the link, then you go to the website and go to inspect and cllick Storage, then you find the cookies, if there are no serviceToken there, you go login to the account.xiaomi.com, then go back to the website from the link you copied then see the cookies back, if you find the serviceToken now you can unlock your phone!