francescotescari/XiaoMiToolV2 #108: Service token error

flixcrls, 2023-08-15:
When the update xiaomitool V2 to fix this, i dont know how to unlock my phone with mac

amitava82, 2023-09-03:
Any solution yet?

flixcrls, 2023-09-03:
@amitava82 nope, if u use intel mac just install vb/vm

vsvsav, 2023-10-09:
> @amitava82 nope, if u use intel mac just install vb/vm

This a permanent problem and for Win10. Global Xiaomi AI bots answered nonsense about their official MiUnlocker utility, it doesn’t work either.

tommasoverolaLD, 2023-10-09:
Did you manage to solve this problem ?

flixcrls, 2023-10-11:
> > @amitava82 nope, if u use intel mac just install vb/vm
> 
> This a permanent problem and for Win10. Global Xiaomi support - AI bots - answered nonsense about their official MiUnlocker utility, it doesn’t work either.
> 
> PS One of the letters was simply answered, go to the service center, they should help you there. In fact, the answer is already that Xiomi believes that you are not the owners of your phone and do not have the right to flash it with your 5 year Mi Account. Is it even worth buying a Xiomi now?
> 
> PS2 Old phones for no reason with a locked bootloader stop loading one day (I researched my local marketplace, a lot of such phones in last time: won't load into fastboot too with locked bootloader with official firmware! but mi flash detects and tries to write, but locked partitions...), and nothing more can be done if you forgot to unlock while the phone was under warranty. (Go to the official service center and pay more than your old phone is probably worth... Throw it away and buy a new one, even if it could be good for your little childrens)
> 
> PS3 I personally won’t buy it again, and they also have a Mi Cloud that allows you to download and delete only one file/note/sms at a time, one file at a time!! (you could not export all, you could not delete all, Apple cloud allowed you to download all, any company allows ) and this is if you managed to backup up, it’s tough.
> 
> PS4 A very long time ago, Sony(-Erricson) had its own service, you insert the phone, press one button Repair, and it restored to factory (it automatically download what it needs). No one else has this., unfortunately.
> 
> [image]

this company is greedy day by day

flixcrls, 2023-10-11:
> This a permanent problem and for Win10. Global Xiaomi support - AI bots - answered nonsense about their official MiUnlocker utility, it doesn’t work either.
>
> PS One of the letters was simply [...]

this company is greedy day by day, and the new redmi note doesnt have micro sd slot, search and buy the new phone for open source and easy to ubl it, for now i uninstall many bloatware in xiaomi shit rom

amitava82, 2023-10-12:
I've already decided not to buy xiomi phones ever again. With Pixel 8 7Y OS update it is tempting or S24 next year :)

flixcrls, 2023-10-12:
> I've already decided not to buy xiomi phones ever again. With Pixel 8 7Y OS update it is tempting or S24 next year :)

yeah, the transsion group(infinix,itel,tecno) has potential but there are not many cusrom, it take times to dev aosp for making cusrom in that phone