francescotescari/XiaoMiToolV2 #111: stuck in fastboot, then error in application

damnms, 2023-09-08:
i tried to unlock my redmi pad on linux fedora 38, all looks good, then the pad reboots and it displays "fastboot", and then the application just does nothing. when running adb devices, i see nothing. when i reboot the pad and do adb devices, i can see it again. not sure if its an application problem or some sort of problem with the pad

1nfameArts, 2023-12-22:
You couldn't see your tablet through adb because it was in Fastboot. Use the appropriate command for fastboot using the `fastboot` command, usually packaged with platform tools. Rebooting the pad just lead it into the mode where ADB is enabled, but fastboot isn't.
Short worded: Your tablet was in fastboot mode, which is compatible only with the `fastboot` command.

damnms, 2023-12-22:
yeah got that working but forgot to close that ticket ;)

closed 2023-12-22 by damnms