francescotescari/XiaoMiToolV2 #114: Login form not work. Login SMS code verification not show a captcha.

EvgeniyBlinov, 2023-11-27:
Login SMS code verification not show a captcha.

riegaz, 2023-12-15:
> @riegaz Try this https://gist.github.com/shvchk/bfcf9b863bb3062138497ecbb6f8860a

This is the error I get.

`parallels@debian-gnu-linux-11:/home/XiaoMiToolV2$ ./gradlew build --debug
2023-12-15T14:02:34.245+0100 [INFO] [org.gradle.internal.nativeintegration.services.NativeServices] Initialized native services in: /home/parallels/.gradle/native
2023-12-15T14:02:34.275+0100 [LIFECYCLE] [org.gradle.launcher.cli.DebugLoggerWarningAction] 
#############################################################################
   WARNING WARNING WARNING WARNING WARNING WARNING WARNING WARNING WARNING

   Debug level logging will leak security sensitive information!

   https://docs.gradle.org/6.5/userguide/logging.html#sec:debug_security
#############################################################################

2023-12-15T14:02:34.321+0100 [DEBUG] [org.gradle.launcher.daemon.client.DaemonClient] Executing build f4db3535-abd3-428a-8429-b6803731458d in daemon client {pid=305478}
2023-12-15T14:02:34.323+0100 [DEBUG] [org.gradle.internal.remote.internal.inet.InetAddresses] Adding IP addresses for network interface enp0s5
2023-12-15T14:02:34.323+0100 [DEBUG] [org.gradle.internal.remote.internal.inet.InetAddresses] Is this a loopback interface? false
2023-12-15T14:02:34.323+0100 [DEBUG] [org.gradle.internal.remote.internal.inet.InetAddresses] Adding remote address /fe80:0:0:0:21c:42ff:fe16:5891%enp0s5
2023-12-15T14:02:34.323+0100 [DEBUG] [org.gradle.internal.remote.internal.inet.InetAddresses] Adding remote address /fdb2:2c26:f4e4:0:21c:42ff:fe16:5891%enp0s5
2023-12-15T14:02:34.323+0100 [DEBUG] [org.gradle.internal.remote.internal.inet.InetAddresses] Adding remote address /fdb2:2c26:f4e4:0:7856:5a32:5053:54fb%enp0s5
2023-12-15T14:02:34.323+0100 [DEBUG] [org.gradle.internal.remote.internal.inet.InetAddresses] Adding remote address /10.211.55.4
2023-12-15T14:02:34.323+0100 [DEBUG] [org.gradle.internal.remote.internal.inet.InetAddresses] Adding IP addresses for network interface lo
2023-12-15T14:02:34.323+0100 [DEBUG] [org.gradle.internal.remote.internal.inet.InetAddresses] Is this a loopback interface? true
2023-12-15T14:02:34.323+0100 [DEBUG] [org.gradle.internal.remote.internal.inet.InetAddresses] Adding loopback address /0:0:0:0:0:0:0:1%lo
2023-12-15T14:02:34.323+0100 [DEBUG] [org.gradle.internal.remote.internal.inet.InetAddresses] Adding loopback address /127.0.0.1
2023-12-15T14:02:34.326+0100 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Waiting to acquire shared lock on daemon addresses registry.
2023-12-15T14:02:34.326+0100 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Lock acquired on daemon addresses registry.
2023-12-15T14:02:34.327+0100 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Releasing lock on daemon addresses registry.
2023-12-15T14:02:34.328+0100 [DEBUG] [org.gradle.internal.remote.internal.inet.TcpOutgoingConnector] Attempting to connect to [b05efc57-e175-459b-af10-742dcc169732 port:40577, addresses:[/127.0.0.1]].
2023-12-15T14:02:34.328+0100 [DEBUG] [org.gradle.internal.remote.internal.inet.TcpOutgoingConnector] Trying to connect to address /127.0.0.1.
2023-12-15T14:02:34.330+0100 [DEBUG] [org.gradle.internal.remote.internal.inet.TcpOutgoingConnector] Connected to address /127.0.0.1:40577.
2023-12-15T14:02:34.334+0100 [DEBUG] [org.gradle.launcher.daemon.client.DaemonClient] Connected to daemon DaemonInfo{pid=305387, address=[b05efc57-e175-459b-af10-742dcc169732 port:40577, addresses:[/127.0.0.1]], state=Idle, lastBusy=1702645335528, context=DefaultDaemonContext[uid=6b875d8b-62ab-4515-aa63-426cea8e2db8,javaHome=/usr/lib/jvm/java-11-openjdk-arm64,daemonRegistryDir=/home/parallels/.gradle/daemon,pid=305387,idleTimeout=10800000,priority=NORMAL,daemonOpts=--add-opens,java.base/java.util=ALL-UNNAMED,--add-opens,java.base/java.lang=ALL-UNNAMED,--add-opens,java.base/java.lang.invoke=ALL-UNNAMED,--add-opens,java.prefs/java.util.prefs=ALL-UNNAMED,-Dfile.encoding=UTF-8,-Duser.country=US,-Duser.language=en,-Duser.variant]}. Dispatching request Build{id=f4db3535-abd3-428a-8429-b6803731458d, currentDir=/home/XiaoMiToolV2}.
2023-12-15T14:02:34.334+0100 [DEBUG] [org.gradle.launcher.daemon.client.DaemonClientConnection] thread 1: dispatching class org.gradle.launcher.daemon.protocol.Build
2023-12-15T14:02:34.339+0100 [DEBUG] [org.gradle.launcher.daemon.client.DaemonClient] Received result org.gradle.launcher.daemon.protocol.BuildStarted@7103cb56 from daemon DaemonInfo{pid=305387, address=[b05efc57-e175-459b-af10-742dcc169732 port:40577, addresses:[/127.0.0.1]], state=Idle, lastBusy=1702645335528, context=DefaultDaemonContext[uid=6b875d8b-62ab-4515-aa63-426cea8e2db8,javaHome=/usr/lib/jvm/java-11-openjdk-arm64,daemonRegistryDir=/home/parallels/.gradle/daemon,pid=305387,idleTimeout=10800000,priority=NORMAL,daemonOpts=--add-opens,java.base/java.util=ALL-UNNAMED,--add-opens,java.base/java.lang=ALL-UNNAMED,--add-opens,java.base/java.lang.invoke=ALL-UNNAMED,--add-opens,java.prefs/java.util.prefs=ALL-UNNAMED,-Dfile.encoding=UTF-8,-Duser.country=US,-Duser.language=en,-Duser.variant]} (build should be starting).
2023-12-15T14:02:34.340+0100 [INFO] [org.gradle.launcher.daemon.server.exec.LogToClient] The client will now receive all logging from the daemon (pid: 305387). The daemon log file: /home/parallels/.gradle/daemon/6.5/daemon-305387.out.log
[... cut ...]
2023-12-15T14:02:34.421+0100 [DEBUG] [org.gradle.launcher.daemon.client.DaemonClientConnection] thread 1: connection stop
2023-12-15T14:02:34.424+0100 [LIFECYCLE] [org.gradle.launcher.cli.DebugLoggerWarningAction] 
#############################################################################
   WARNING WARNING WARNING WARNING WARNING WARNING WARNING WARNING WARNING

   Debug level logging will leak security sensitive information!

   https://docs.gradle.org/6.5/userguide/logging.html#sec:debug_security
#############################################################################`

riegaz, 2023-12-15:
I try to run it on an Macbook M2. Could it be that arm64 isn't supported?