francescotescari/XiaoMiToolV2 #115: Download failed version global

DimitriTriantafilo, 2023-12-06:
[image]

experienced this error when downloading global version for xiaomi redmi note 8

vladimir-kuznetsov, 2024-02-15:
You can download firmware manually and flash it using ADB sideload:
https://www.xiaomitool.com/adb

juniosxs, 2024-02-16:
> You can download firmware manually and flash it using ADB sideload: https://www.xiaomitool.com/adb

how to do this with locked bootloader?

vladimir-kuznetsov, 2024-02-17:
> > You can download firmware manually and flash it using ADB sideload: https://www.xiaomitool.com/adb
> 
> how to do this with locked bootloader?

This tool uses adb sideload and can't help in case of locked bootloader.