francescotescari/XiaoMiToolV2 #118: Download error - Illegal char <?>

SU-R27, 2024-02-04:
Hi,
Im getting a download error when trying to recover a Xiaomi 11 Lite.

[image]

vladimir-kuznetsov, 2024-02-12:
Hi, I have the same error for Redmi 7A
[image]

vladimir-kuznetsov, 2024-02-15:
So issue solved using ADB sideload
https://www.xiaomitool.com/adb

juniosxs, 2024-02-16:
> So issue solved using ADB sideload https://www.xiaomitool.com/adb

how to do this? Does it work with locked bootloader?

vladimir-kuznetsov, 2024-02-17:
> > So issue solved using ADB sideload https://www.xiaomitool.com/adb
> 
> how to do this? Does it work with locked bootloader?

If you need to flash an original firmware it will do the job. Just download proper version.

ShreyansJa1n, 2024-02-18:
> > > So issue solved using ADB sideload https://www.xiaomitool.com/adb
> > 
> > 
> > how to do this? Does it work with locked bootloader?
> 
> If you need to flash an original firmware it will do the job. Just download proper version.

@vladimir-kuznetsov, I have downloaded the correct Full-ROM (OTA) package and when I use XiaomiADB, sideload starts and ends within a millisecond with error "ERROR: Sideload error -> adb: failed to read command: no error".

If possible, can you help me with that? Thanks.

vladimir-kuznetsov, 2024-02-18:
> @vladimir-kuznetsov, I have downloaded the correct Full-ROM (OTA) package and when I use XiaomiADB, sideload starts and ends within a millisecond with error "ERROR: Sideload error -> adb: failed to [...]

You need to switch your phone into recovery mode (hold Volume Up during reboot) and select " connect to mi assistant". Then connect to PC and run adb sideload. In my case it reported some issues but flashing started. The phone should reboots by itself after everything completed.
If this does not help I gues you need to try Emergency recovery but I never try it.

InnaeKD, 2024-12-30:
> You need to switch your phone into recovery mode (hold Volume Up during reboot) and select " connect to mi assistant". Then connect to PC and run adb sideload. In my case it reported some issues but [...]

Hi M. do i need to download the recovery rom ? cause when i try with the fastboot one it says cannot install unofficial rom

Fantoooc, 2025-05-23:
[Installable(1).txt](https://github.com/user-attachments/files/20412034/Installable.1.txt)

[Installable(2).txt](https://github.com/user-attachments/files/20412784/Installable.2.txt)

To fix this problem you need to modify the Installable.java file, I did it half through a neural network, you can not blame me for the crooked code. If you do not know Java at all, you can replace the file with Installable(1).txt, while renaming it to Installable.java, but you may have a slow download speed.

If the download speed is too low, try downloading the package using aria2, then uploading it to your Google Drive. Open access to it via the link -> Get download link(Copy file link -> paste in your browser -> go to download page(screenshot *download page) -> Start downloading from your Gdrive and copy download link from browser (right mouse click)) -> Replace my link with yours, replace part of the link with yours in line 160 (The screenshot shows where you need to replace). (You need file Installable(2).txt, rename it to Installable.java)

*instructions on how to download the update package via aria2 will be published below

[image]

[image]

*download page

[image]

[image]

* translated with google😒

Fantoooc, 2025-05-23:
Your link should look something like this:

http://bigota.d.miui.com/V12.5.2.0.QCMRUXM/miui_PINERUGlobal_V12.5.2.0.QCMRUXM_c423a86604_10.0.zip

http://bigota.d.miui.com/version/region_version_some random symbols.zip

Your error contains a file name. Rewrite it after the fourth slash and after the third insert the version number of your firmware, I showed how it looks above. This will be your link.

Next, download the aria2 package, I think you will figure out how to do it. After that, open the console and write
aria2c -x 16 -s 16 -k 4M --file-allocation=none "your link"
let's say, for me it looks like this:
aria2c -x 16 -s 16 -k 4M --file-allocation=none "http://bigota.d.miui.com/V12.5.2.0.QCMRUXM/miui_PINERUGlobal_V12.5.2.0.QCMRUXM_c423a86604_10.0.zip"

After the download is complete, the firmware file will appear in your downloads folder (probably). You can continue to follow the instructions in the previous post.

Fantoooc, 2025-05-23:
I really hope that I helped someone. If there are questions, if I notice - I will try to answer.

Fantoooc, 2025-09-25:
> For me the download is taking a hell lot of time. Is it possible to replace an already downloaded to the tmp folder (I'm using Linux deb) or some other workaround?

idk, i did it by downloading file using aria2 & re-upload it on my gdrive