francescotescari/XiaoMiToolV2 #132: Download Task Failed (I need help pls)

T3BB3L, 2024-09-26:
Hi, I'm asking to whoever who read this for help. Well, my phone got bricked 1 year and a half ago, I was using normally when out of nowhere it began to restarting again and again, back in time I tried everything and eventually I got tired, but now i'm trying to revive my old Poco M3 pro 5g, so I downloaded and installed XiaoMiToolV2 but this error occurs

[image]

I read on issues section and also tried ADB, dragging rom and opening adb.exe, but also this error occurs

[image]

Yes, bootloader is blocked and I don't know how to continue, I need help, TY :)

InnaeKD, 2024-12-30:
hi bro i'm dealing with the same problem but mine is a bit different. You have to turn your phone on recovery mode by pressing power and vol up and enter connect with mi assistant then use xiaomi adb

InnaeKD, 2024-12-30:
> Hi, I'm asking to whoever who read this for help. Well, my phone got bricked 1 year and a half ago, I was using normally when out of nowhere it began to restarting again and again, back in time I tried everything and eventually I got tired, but now i'm trying to revive my old Poco M3 pro 5g, so I downloaded and installed XiaoMiToolV2 but this error occurs
> 
> ![GITHUB1](https://private-user-images.githubusercontent.com/182998650/371271061-20da1cef-6890-4378-8b3d-8523e8f20aea.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU1Njk0OTYsIm5iZiI6MTczNTU2OTE5NiwicGF0aCI6Ii8xODI5OTg2NTAvMzcxMjcxMDYxLTIwZGExY2VmLTY4OTAtNDM3OC04YjNkLTg1MjNlOGYyMGFlYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIzMFQxNDMzMTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xNzljMDgxZTA5MzEyMjMyMTA3ZmY1ZWM0NTdlYWFhNDEwNTViZTI3MzE5YzNmMmFhYjlmNzcxMzViOWMwYTkxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.1depvMbPbKgeHEEVj36zbQ1PhIsZqICFQTL-L8kId4c)
> 
> I read on issues section and also tried ADB, dragging rom and opening adb.exe, but also this error occurs
> 
> ![GITHUB2](https://private-user-images.githubusercontent.com/182998650/371271526-890d686b-c3f8-4543-b575-67eb2785c095.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU1Njk0OTYsIm5iZiI6MTczNTU2OTE5NiwicGF0aCI6Ii8xODI5OTg2NTAvMzcxMjcxNTI2LTg5MGQ2ODZiLWMzZjgtNDU0My1iNTc1LTY3ZWIyNzg1YzA5NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIzMFQxNDMzMTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xYzQ3ODgyNDcyMTg5NzM3ZmU3ODczYjFhZThkMzAwNjQ1ZWQyNzk4OTdmYzYzNTk1MjMwYmU1OWM2NGEwYTZjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.g95q4YMqYsPD47AQF6xrDuQPOm-YKrKz76CYU7shHXE)
> 
> Yes, bootloader is blocked and I don't know how to continue, I need helpyou

> Hi, I'm asking to whoever who read this for help. Well, my phone got bricked 1 year and a half ago, I was using normally when out of nowhere it began to restarting again and again, back in time I tried everything and eventually I got tired, but now i'm trying to revive my old Poco M3 pro 5g, so I downloaded and installed XiaoMiToolV2 but this error occurs
> 
> ![GITHUB1](https://private-user-images.githubusercontent.com/182998650/371271061-20da1cef-6890-4378-8b3d-8523e8f20aea.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU1Njk0OTYsIm5iZiI6MTczNTU2OTE5NiwicGF0aCI6Ii8xODI5OTg2NTAvMzcxMjcxMDYxLTIwZGExY2VmLTY4OTAtNDM3OC04YjNkLTg1MjNlOGYyMGFlYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIzMFQxNDMzMTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xNzljMDgxZTA5MzEyMjMyMTA3ZmY1ZWM0NTdlYWFhNDEwNTViZTI3MzE5YzNmMmFhYjlmNzcxMzViOWMwYTkxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.1depvMbPbKgeHEEVj36zbQ1PhIsZqICFQTL-L8kId4c)
> 
> I read on issues section and also tried ADB, dragging rom and opening adb.exe, but also this error occurs
> 
> ![GITHUB2](https://private-user-images.githubusercontent.com/182998650/371271526-890d686b-c3f8-4543-b575-67eb2785c095.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU1Njk0OTYsIm5iZiI6MTczNTU2OTE5NiwicGF0aCI6Ii8xODI5OTg2NTAvMzcxMjcxNTI2LTg5MGQ2ODZiLWMzZjgtNDU0My1iNTc1LTY3ZWIyNzg1YzA5NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIzMFQxNDMzMTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xYzQ3ODgyNDcyMTg5NzM3ZmU3ODczYjFhZThkMzAwNjQ1ZWQyNzk4OTdmYzYzNTk1MjMwYmU1OWM2NGEwYTZjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.g95q4YMqYsPD47AQF6xrDuQPOm-YKrKz76CYU7shHXE)
> 
> Yes, bootloader is blocked and I don't know how to continue, I need help, TY :)

You have to choose your device in xmt2 , in xiaomi adb you have to turn your phone on recovery mode by pressing vol up and power key then go to connect with mi assistant

T3BB3L, 2024-12-30:
Hi, thanks for your response. I was finally able to unlock it with a Chinese app, I have no idea how it worked, but it did haha. This happened by connecting it to Mi assistant and the app on the computer did all the work, automatically searched for a rom online and reinstalled it. Cheers.

Pd: This was a few months ago, the phone works perfectly but the bug happens again when I turn it off or when the battery runs out, although I manage to get it out of the bug with recovery mode.

InnaeKD, 2024-12-31:
Hey please can you share this app with me ? i'm facing this problem

T3BB3L, 2024-12-31:
Obviously, I'll send it to you without any problem. But let me look for the app on the computer, I don't remember the name. Nor do I remember how I found it. Before that I had tried absolutely everything.

T3BB3L, 2024-12-31:
I found it, it's called MiPhoneAssistant, but I don't remember the website. But I do remember it was a Chinese website, so I had to use the translator with my phone's camera. I recommend you install a version, and if it doesn't work or there is a problem, uninstall it and try another version from another page. Only 4.2.1028.10be worked for me. 

Maybe it's this page: xiaomitools.com/download/mi-pc-suite-miassistant-v4-2-1028-10/

Or this other one: miuiver.com/mi-phone-assistant/

Or this, this one actually sounds familiar to me: https://web.vip.miui.com/page/info/mio/mio/detail?app_version=dev.20051&postId=37066100

I hope you can solve it, if not, I'm here to help.

InnaeKD, 2024-12-31:
thank you M. may God bless you for your help

VinithSamson11, 2025-02-10:
I did download the 12.5 version, later got this error

[image]

here is the downloaded file. 

[image]

When I tried to reselect the package from mi-assistance-rom folder its not recognized. (I think it requires file in .zip type).

could u please help me @T3BB3L @InnaeKD

VinithSamson11, 2025-02-10:
@T3BB3L could u share that 3GB fastboot file, where mi-assistance downloads. Please.

I just changed the file type to zip, it worked. but mi-assistance didn't recognize the zip file as official.

[image]

T3BB3L, 2025-02-10:
Tell me a little more so I understand. I'm not an expert in these matters, it was just that my phone was bricked 😄

VinithSamson11, 2025-02-10:
The package was being downloaded through the Mi-Assistant software. Once the download was completed, I encountered this error (refer to the first image in this comment: https://github.com/francescotescari/XiaoMiToolV2/issues/132#issuecomment-2647528577). I then started searching for the downloaded package's location, and while doing so, I found the package and noticed there was a mistake in the file type (see the second image in the comment: https://github.com/francescotescari/XiaoMiToolV2/issues/132#issuecomment-2647528577). I tried to manually load the file, but the Mi-Assistant software didn't recognize it. So, I edited the package file extension to ".zip."

The entire package could be extracted without any issues, but Mi-Assistant still shows it as an unofficial package (even though the package was originally downloaded by the Mi-Assistant software itself).