francescotescari/XiaoMiToolV2 #134: when i want to unlock the bootloader i get error

h-lavaei, 2025-01-12:
error : Xiaomi procedure failed: [getServiceToken] Missing serviceToken cookie
phone: Redmi note 10 pro

clivinn-shla81092, 2025-01-12:
Hello, we tried to solve the issue.

This is what we did:

Modified the XiaomiKeystore class to properly set and retrieve the serviceToken cookie. Added a new method to set the serviceToken and updated the requireServiceKeyAndToken method to use it.

You can review changes in this commit: https://github.com/clivinn-shla81092/francescotescari-XiaoMiToolV2-134/commit/b11af1a169184a428b04aea66d2c3f09609f8f66.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

kaliiiiiiiiii, 2025-03-25:
Try https://github.com/francescotescari/XiaoMiToolV2/issues/23#issuecomment-2741124790

ebym, 2025-05-19:
Same here, I have token from https://github.com/Gtajisan/MiUnlockTool and this thing don't work when trying to unlock, show error with python [ FileNotFoundError: [Errno 2] No such file or directory: 'python' ]